rogerclarkmelbourne/Arduino_STM32 #579: incorrect core headers for F4 version

smarq8, 2018-11-30:
I begin use STM32F405 with this library for arduino and after few days trying to write own library for SPI because of some unexpected behaviour I probably found incoherence inside core headers.

```
// file: STM32F4\cores\maple\libmaple\rccF4.h
/* APB2 peripheral reset register */
#define RCC_APB2RSTR_TIM11RST_BIT       21
#define RCC_APB2RSTR_TIM10RST_BIT       20
#define RCC_APB2RSTR_TIM9RST_BIT        19
#define RCC_APB2RSTR_ADC3RST_BIT        15
#define RCC_APB2RSTR_USART1RST_BIT      14
#define RCC_APB2RSTR_TIM8RST_BIT        13
#define RCC_APB2RSTR_SPI1RST_BIT        12
#define RCC_APB2RSTR_TIM1RST_BIT        11
#define RCC_APB2RSTR_ADC2RST_BIT        10
#define RCC_APB2RSTR_ADC1RST_BIT        9
#define RCC_APB2RSTR_IOPGRST_BIT        8
#define RCC_APB2RSTR_IOPFRST_BIT        7
#define RCC_APB2RSTR_IOPERST_BIT        6
#define RCC_APB2RSTR_IOPDRST_BIT        5
#define RCC_APB2RSTR_IOPCRST_BIT        4
#define RCC_APB2RSTR_IOPBRST_BIT        3
#define RCC_APB2RSTR_IOPARST_BIT        2
#define RCC_APB2RSTR_AFIORST_BIT        0

/* APB1 peripheral reset register */

#define RCC_APB1RSTR_DACRST_BIT         29
#define RCC_APB1RSTR_PWRRST_BIT         28
#define RCC_APB1RSTR_BKPRST_BIT         27
#define RCC_APB1RSTR_CANRST_BIT         25
#define RCC_APB1RSTR_USBRST_BIT         23
#define RCC_APB1RSTR_I2C2RST_BIT        22
#define RCC_APB1RSTR_I2C1RST_BIT        21
#define RCC_APB1RSTR_UART5RST_BIT       20
#define RCC_APB1RSTR_UART4RST_BIT       19
#define RCC_APB1RSTR_USART3RST_BIT      18
#define RCC_APB1RSTR_USART2RST_BIT      17
#define RCC_APB1RSTR_SPI3RST_BIT        15
#define RCC_APB1RSTR_SPI2RST_BIT        14
#define RCC_APB1RSTR_WWDRST_BIT         11
#define RCC_APB1RSTR_TIM14RST_BIT       8
#define RCC_APB1RSTR_TIM13RST_BIT       7
#define RCC_APB1RSTR_TIM12RST_BIT       6
#define RCC_APB1RSTR_TIM7RST_BIT        5
#define RCC_APB1RSTR_TIM6RST_BIT        4
#define RCC_APB1RSTR_TIM5RST_BIT        3
#define RCC_APB1RSTR_TIM4RST_BIT        2
#define RCC_APB1RSTR_TIM3RST_BIT        1
#define RCC_APB1RSTR_TIM2RST_BIT        0
```
and now just compare with reference manual RM0090

[image]
[image]

or with "stm32f4xx.h"
```
/********************  Bit definition for RCC_APB2RSTR register  **************/
#define  RCC_APB2RSTR_TIM1RST                ((uint32_t)0x00000001) // 0
#define  RCC_APB2RSTR_TIM8RST                ((uint32_t)0x00000002) // 1
#define  RCC_APB2RSTR_USART1RST              ((uint32_t)0x00000010) // 4
#define  RCC_APB2RSTR_USART6RST              ((uint32_t)0x00000020) // 5
#define  RCC_APB2RSTR_ADCRST                 ((uint32_t)0x00000100) // 8
#define  RCC_APB2RSTR_SDIORST                ((uint32_t)0x00000800) // 11
#define  RCC_APB2RSTR_SPI1RST                ((uint32_t)0x00001000) // 12
#define  RCC_APB2RSTR_SPI4RST                ((uint32_t)0x00002000) // 13
#define  RCC_APB2RSTR_SYSCFGRST              ((uint32_t)0x00004000) // 14
#define  RCC_APB2RSTR_TIM9RST                ((uint32_t)0x00010000) // 16
#define  RCC_APB2RSTR_TIM10RST               ((uint32_t)0x00020000) // 17
#define  RCC_APB2RSTR_TIM11RST               ((uint32_t)0x00040000) // 18
#define  RCC_APB2RSTR_SPI5RST                ((uint32_t)0x00100000) // 20
#define  RCC_APB2RSTR_SPI6RST                ((uint32_t)0x00200000) // 21
```
As you can see for example TIM11RST_BIT==21 while it should be 18!
It doesn't look like literal error but as writing library with incorrect sources.
I randomly checked few other 'things' with similar resoult.
Then I miss something or this library for F4 is untrustworthy anymore for me.

stevstrong, 2018-12-01:
This and many more is corrected in my repo, check the forum.

smarq8, 2018-12-01:
Maybe "many more" is fixed, I do not know. However "this" still not.

random check from your repo - fail
https://github.com/stevstrong/Arduino_STM32
[image]

APB2 peripheral reset register exactly same as in 1st post (incorrect).
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/libmaple/rccF4.h
```
/* APB2 peripheral reset register */

#define RCC_APB2RSTR_TIM11RST_BIT       21
#define RCC_APB2RSTR_TIM10RST_BIT       20
#define RCC_APB2RSTR_TIM9RST_BIT        19
#define RCC_APB2RSTR_ADC3RST_BIT        15
#define RCC_APB2RSTR_USART1RST_BIT      14
#define RCC_APB2RSTR_TIM8RST_BIT        13
#define RCC_APB2RSTR_SPI1RST_BIT        12
#define RCC_APB2RSTR_TIM1RST_BIT        11
#define RCC_APB2RSTR_ADC2RST_BIT        10
#define RCC_APB2RSTR_ADC1RST_BIT        9
#define RCC_APB2RSTR_IOPGRST_BIT        8
#define RCC_APB2RSTR_IOPFRST_BIT        7
#define RCC_APB2RSTR_IOPERST_BIT        6
#define RCC_APB2RSTR_IOPDRST_BIT        5
#define RCC_APB2RSTR_IOPCRST_BIT        4
#define RCC_APB2RSTR_IOPBRST_BIT        3
#define RCC_APB2RSTR_IOPARST_BIT        2
#define RCC_APB2RSTR_AFIORST_BIT        0
```

stevstrong, 2018-12-01:
Is this different from F407? Because at this time only F407 is supported.

smarq8, 2018-12-01:
According to RM0090 then in APB2RSTR case 405 and 407 are same, there is difference but to 42xxx/43xxx and with backward compatibility, I also checked other 4xx reference manuals with same result. As I said, I am almost sure current defines are imported/writed from unknown source wchith is not compatible for destined uC of this library.

[image]
[image]

smarq8, 2018-12-01:
Unfortunately not because I only catch this accidentally and there may be a lot of work to verify all files.

stevstrong, 2019-09-15:
The F4 branch of this repo is currently not supported.
Instead, I would recommend either my repo or the official STM repo.
Anyway, if you find more such wrong bit definitions I would be happy if you could open an issue in my repo. Thanks in advance.

stevstrong, 2019-10-26:
I will close this because it is not relevant to the current status where that header file is actually not used.

closed 2019-10-26 by stevstrong