rogerclarkmelbourne/Arduino_STM32 #582: Command-line builds not working

mithat, 2018-12-10:
I'm running Arduino 1.8.8 under Linux (Debian sid). I'm trying to configure a command-line build process for stm32duino projects using the `arduino` command line interface. It seems stm32duino isn't handling this properly. In particular:

### Specifying a build directory

When I issue a command to build into a specific directory:

```
/path/to/arduino --board stm32duino:STM32F1:genericSTM32F103C:cpu_speed=speed_72mhz,opt=osstd --pref build.path=release --verify build-test.ino
```

things seem to build as expected the first time.:

```
$ ls -1 release
build.options.json
build-test.ino.bin
build-test.ino.elf
build-test.ino.map
core
includes.cache
libraries
preproc
sketch
```

But on subsequent builds, thing go wonky. The compile fails:

```
Build options changed, rebuilding all
build-test:1:20: error: config.h: No such file or directory
 #include "config.h"
                    ^
compilation terminated.
exit status 1
```

and the contents of the build directory become:

```
$ ls -1 release
build.options.json
sketch
```

The file `build-test.ino` does indeed `#include config.h`. The rebuild can't seem to find it.

### Not specifying a build directory

When I issue a command that doesn't build into a specific directory:

```
/path/to/arduino --board stm32duino:STM32F1:genericSTM32F103C:cpu_speed=speed_72mhz,opt=osstd --verify build-test.ino
```

things are built into a directory under `/tmp`, and when the build is complete, that directory vanishes.

Wiping of the `/tmp` directory is probably expected behavior with the CLI. (The IDE maintains the directory until you close it.) I mention it here only to point out that not specifying a build directory isn't a workaround.

### Sources
[build-test.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/2664909/build-test.zip)

rogerclarkmelbourne, 2018-12-10:
This repo only works with the Arduino IDE

Please feel free to submit a PR to add the functionality to support command line builds

per1234, 2018-12-11:
There's a similar report here (except not using Arduino_STM32): https://github.com/arduino/Arduino/issues/8271

fpistm, 2018-12-11:
FYI, I've tested build using arduino-builder with your core and it was functional.

rogerclarkmelbourne, 2018-12-11:
@fpistm 

Thanks.

fpistm, 2018-12-11:
Just for reference:
http://stm32duino.com/viewtopic.php?f=18&t=3944

mithat, 2018-12-11:
This does appear to be an Arduino problem. Per the [report](https://github.com/arduino/Arduino/issues/8271) @per1234  linked to in [this reply](https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/582#issuecomment-446090750), I tried building/rebuilding to a directory outside the sketch directory using Arduino 1.8.8, and things worked as expected, even on a fairly complex (20+ source files) project.

I also tried building/rebuilding a simple two source file project to a directory inside the sketch directory using Arduino 1.8.5, and that worked fine too. However, it it failed with 1.8.8.

Please feel free to mark this closed.

rogerclarkmelbourne, 2019-01-28:
Closed as this is not a problem with this core

closed 2019-01-28 by rogerclarkmelbourne