rogerclarkmelbourne/Arduino_STM32 #584: ST-link upload error caused by error in /tools/win/stlink_upload.bat

semicolonTransistor, 2018-12-23:
In line 9 of stlink_upload.bat the 4th command line argument is read into the variable str
`set str=%4`
Which is further processed and passed into STLink_CLI as the path to the bin file. However acturally the path to the bin file is passed in as the 3rd command line argument.  This error resulted in the bat file reading in an empty string as path and passing an incorrect path to the STLink_CLI which resulted in the STLink_CLI output the following error in the Arduino IDE
`Loading file...
Unable to open file!`
Changing the line to `set str=%3` seems to fix the issue.

rogerclarkmelbourne, 2019-01-30:
Strange

The bat file works fine for me.

Perhaps this is an issue with the ST-Link CLI having a different arg list in different versions of the CLS

What version are you running ?

fpistm, 2019-01-30:
I already saw this issue. This is probably because you did not select a COM port.
`tools.stlink_upload.upload.pattern="{path}/{cmd}" {serial.port.file} {upload.altID} {upload.usbID} "{build.path}/{build.project_name}.bin"`
Then `{serial.port.file}` is empty which explain why arg number is not properly processed

semicolonTransistor, 2019-01-30:
> Then {serial.port.file} is empty which explain why arg number is not properly processed

That is probably the problem, an empty argument is going to shift everything forward by one. I am also a bit confused as to why all of these parameters were passed to the stlink uploader since only the file path was ever used. Anayways,thanks for your time, I'll closes the issure.

rogerclarkmelbourne, 2019-01-30:
@fpistm 
Thanks

This is a bug in platform.txt

I'll remove {serial.port.file} from the argument list, and also probably remove the upload.altID and upload.usbID as they are both redundant (and legacy items as this command line was probably copied by me from the bootloader upload, and I forgot to remove the redundant arguments

rogerclarkmelbourne, 2019-01-30:
Please retest using the development branch of the repo

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/development

semicolonTransistor, 2019-01-30:
I'll test this as soon as I get my new stlink. My old one died a few days ago. I'll get back to you tomorrow.

rogerclarkmelbourne, 2019-01-30:
OK

No worries

Note. I also updated the STLink CLI program, and I also noticed that it was hanging after the upload completed, so I've added the -NoPrompt argument, so that it does not wait for return to be pressed after upload before the STLink CLS terminates.

I've tested it with my STLink and it seems to work OK now.

rogerclarkmelbourne, 2019-02-04:
FYI
I've copied this fix to the Master branch.

closed 2019-02-04 by rogerclarkmelbourne

ScottLawsonEng1, 2019-12-22:
This is still a problem as downloaded a clean install tonight and replicated the issue, following installation on 3 machines amongst peers whom initially highlighted the same issue.

Selecting the 3rd argument as the string path option allows correct upload. Attempts to remove the additionally passed in "{upload.altID} {upload.usbID}" were made to ensure that the argument index was always correct but though this has been removed from the platform.txt call, these arguments are still passed to the BAT file when it is being called from somewhere.

stevstrong2, 2019-12-22:
Which OS?
I tried under Windows and it works correctly.
Please insert your Arduino build message here.

ScottLawsonEng1, 2019-12-22:
This is running on a Windows 10 Pro Machine with all updates installed, using a fresh install of the latest version of Arduino using the offline installer (1.8.10), two of the other machines are Windows 10 Home and One is Windows 7 Pro which have never had Arduino on them before. All configured for STM32 by downloading this repository as is using clone, and then installed in the hardware directory as per instructions. Okay, build messages below, I turned on verbose output for debug purposes - Only showing the upload sections.

Using set "str=%1" on line 9 of stlink_upload.bat: 

`Sketch uses 14780 bytes (22%) of program storage space. Maximum is 65536 bytes.
Global variables use 3128 bytes (15%) of dynamic memory, leaving 17352 bytes for local variables. Maximum is 20480 bytes.

C:\Users\["User Name"]\Documents\Arduino\hardware\Arduino_STM32/tools/win/stlink_upload.bat {upload.altID} {upload.usbID} C:\Users\["User Name"]\AppData\Local\Temp\arduino_build_616438/STM32_Bluepill_-_Blink_Demo.ino.bin 

C:\Program Files (x86)\Arduino>rem @echo off 
C:\Program Files (x86)\Arduino>rem: Note C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win\ get path of this batch file 
C:\Program Files (x86)\Arduino>rem: Need to change drive if My Documents is on a drive other than C: 
C:\Program Files (x86)\Arduino>set driverLetter=C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win\ 
C:\Program Files (x86)\Arduino>set driverLetter=C: 
C:\Program Files (x86)\Arduino>C:
C:\Program Files (x86)\Arduino>cd C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win\ 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem: the two line below are needed to fix path issues with incorrect slashes before the bin file name 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>set str={upload.altID} 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>set str={upload.altID} 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem: ------------- use STLINK CLI 

C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>stlink\ST-LINK_CLI.exe -c SWD -P {upload.altID} 0x8000000 -Rst -Run -NoPrompt 
STM32 ST-LINK CLI v3.3.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: 262B09013212354D434B4E00
ST-LINK Firmware version: V2J33S7 (Need Update)
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.2 V
Connection mode: Normal
Reset mode: Software reset
Device ID: 0x410 
Device flash Size: 64 Kbytes
Device family: STM32F10xx Medium-density

Loading file...
Unable to open file!

MCU Reset.

Application started.

C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem: Using the open source texane-stlink instead of the proprietary STM stlink exe 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem:texane-stlink\st-flash.exe write {upload.altID} 0x8000000 `

###########################################################################
Using set "str=%3" on line 9 of stlink_upload.bat: 

`Sketch uses 14780 bytes (22%) of program storage space. Maximum is 65536 bytes.
Global variables use 3128 bytes (15%) of dynamic memory, leaving 17352 bytes for local variables. Maximum is 20480 bytes.
C:\Users\["User Name"]\Documents\Arduino\hardware\Arduino_STM32/tools/win/stlink_upload.bat {upload.altID} {upload.usbID} C:\Users\["User Name"]\AppData\Local\Temp\arduino_build_616438/STM32_Bluepill_-_Blink_Demo.ino.bin 

C:\Program Files (x86)\Arduino>rem @echo off 
C:\Program Files (x86)\Arduino>rem: Note C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win\ get path of this batch file 
C:\Program Files (x86)\Arduino>rem: Need to change drive if My Documents is on a drive other than C: 
C:\Program Files (x86)\Arduino>set driverLetter=C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win\ 
C:\Program Files (x86)\Arduino>set driverLetter=C: 
C:\Program Files (x86)\Arduino>C:
C:\Program Files (x86)\Arduino>cd C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win\ 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem: the two line below are needed to fix path issues with incorrect slashes before the bin file name 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>set str=C:\Users\["User Name"]\AppData\Local\Temp\arduino_build_616438/STM32_Bluepill_-_Blink_Demo.ino.bin 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>set str=C:\Users\["User Name"]\AppData\Local\Temp\arduino_build_616438\STM32_Bluepill_-_Blink_Demo.ino.bin 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem: ------------- use STLINK CLI 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>stlink\ST-LINK_CLI.exe -c SWD -P C:\Users\["User Name"]\AppData\Local\Temp\arduino_build_616438\STM32_Bluepill_-_Blink_Demo.ino.bin 0x8000000 -Rst -Run -NoPrompt 
STM32 ST-LINK CLI v3.3.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: 262B09013212354D434B4E00
ST-LINK Firmware version: V2J33S7 (Need Update)
[... 3 lines cut ...]
Connection mode: Normal
Reset mode: Software reset
Device ID: 0x410 
Device flash Size: 64 Kbytes
Device family: STM32F10xx Medium-density

Loading file...
Flash Programming:
  File : C:\Users\["User Name"]\AppData\Local\Temp\arduino_build_616438\STM32_Bluepill_-_Blink_Demo.ino.bin
[... cut ...]
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem: Using the open source texane-stlink instead of the proprietary STM stlink exe 
C:\Users\["User Name"]\Documents\Arduino\Hardware\Arduino_STM32\tools\win>rem:texane-stlink\st-flash.exe write C:\Users\["User Name"]\AppData\Local\Temp\arduino_build_616438\STM32_Bluepill_-_Blink_Demo.ino.bin 0x8000000 
`

In all cases no Serial devices are installed as we don't use that method in our development environment and generally do not have any other serial devices installed, so no devices available to select as the 'dummy' device as some had suggested previously.

stevstrong, 2019-12-23:
Check line 146 of your "platform.txt" file.
It should be like this: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/platform.txt#L146

ScottLawsonEng1, 2019-12-23:
Hi stevstrong,

You are correct, that's where the discrepancy was. Not sure how I missed this as I did edit this line previously, but must have lost track of which combination of edits in each file and thought I had reverted back to the original state of the repository. Apologies for the false alarm, and thanks for your assistance.