rogerclarkmelbourne/Arduino_STM32 #585: 45-maple.rules for linux64

yoh-there, 2019-01-02:
I can confirm that for my 18.04 (linux 64) system, I also needed to add the famous 

, ENV{ID_MM_DEVICE_IGNORE}="1"

as mentioned here https://www.stm32duino.com/viewtopic.php?t=1103&start=40#p26620

I agree the repo should be updated to reflect that change in the linux64 directory, as it already has in the linux one.

Thanks!

stevstrong, 2019-09-16:
I would like to know what kind of changes are necessary to handle this.
The link does not exist anymore so I cannot figure out what to do.
Can anyone help on this?

yoh-there, 2019-09-16:
Go here for the original thread https://web.archive.org/web/20190316225313/https://www.stm32duino.com/viewtopic.php?t=1103&start=40

stevstrong, 2019-09-16:
@yoh-there 
can you please provide a PR?

yoh-there, 2019-09-16:
Sorry, don'tknow what you are asking for. And it's been ages (yup,  month) ago since I fiddles with this. It's all very, very hazy

stevstrong, 2019-09-16:
I was asking for a pull request.
And I know it's been a long time ago, but all things should come to an end.

yoh-there, 2019-09-16:
Oh I am sorry, I never fiddled in the repository, I just made the changes after installing in the rules.

stevstrong, 2019-09-16:
Ok, I see.
So there is no change to be made in the repo, right?
In this case we could close this issue?

yoh-there, 2019-09-16:
Not by me no. I was basically confirming that the proposed change as described in the forums was, IMHO, a good idea, and that here would be a better place to track it than some forum posts. As far as I am concerned, closing is fine, which I'll do right now. At least the issue is documented.

closed 2019-09-16 by yoh-there

stevstrong, 2019-09-16:
Possible duplicate of #651