rogerclarkmelbourne/Arduino_STM32 #598: Suggestion of new instructions for windows drivers installation

FCam1, 2019-02-12:
Hi, 
I met an error trying to install drivers on windows 7. When I launched `install_drivers.bat` the program started but failed because "an other programm is already trying to install the driver".  In fact windows tried, at the same time and each time to install the driver from his data base. 
To avoid this problem I desactivated temporary the auto set up. 
Go to : _control panel>system and security>System_
On the left  choose _System protection_ > _Material device installation parameter_ then select the _manually option_ then choose to _Never install drivers_ and  _Save the configuration_
After that launch `install_drivers.bat` , when it's done return in the configuration and cancel the previous changes. 
I don't have an English version of windows so the text is as close as possible.
Maybe it is also possible to apply this method for only one device and permanently but I didn't look for.
I think it could be nice to add these lines in the wiki for the windows users.

stevstrong, 2019-09-15:
I don't think that this is a "normal" scenario when installing the driver.
When I installed them on my Win7 PC, I had no such issues, so I think your issue is related only to your system.
What is not mentioned in the wiki is that you should run the batch file with administrator rights.

rogerclarkmelbourne, 2020-02-06:
Closed because of no response from the OP

closed 2020-02-06 by rogerclarkmelbourne