rogerclarkmelbourne/Arduino_STM32 #599: USB can not reenumerate

l15y, 2019-02-18:
Here is my code
```
void setup() {
  pinMode(PA12, OUTPUT_OPEN_DRAIN); // use external 1k5 resistor
  digitalWrite(PA12, LOW);  // force low to signal reset
  delay(50);
  digitalWrite(PA12, HIGH); // actually floats high, not driven
  USBComposite.setVendorId(0x6666);
  USBComposite.setProductId(0x0001);
  HID.begin(CompositeSerial, HID_JOYSTICK);
}
```

rogerclarkmelbourne, 2019-02-18:
Please explain the issue.

I think you are trying for force re-enumeration by driving PA12 low for 50mS. But the timing of this method is critical, and 50mS may be either too long or too short.

I don't know the value used in the code, you'd need to check that yourself, and possibly replicate the code exactly as the core does when it forces re-enumeration in usb_serial.cpp

l15y, 2019-02-18:
Yes,I'm trying for force re-enumeration by driving PA12 low for 50mS.
I don't know wether the time is wrong.

After I use ST-Link download program,my code can't re-enumeration,even if I press the reset button.
But when I re-plug USB,it runs well.Then I press the reset button many times,each of then can re-enumeration.(My BluePill is powerd only by USB,not ST-Link)

rogerclarkmelbourne, 2019-02-18:
See
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/b5cd37696b7057f8489c0a17801420756b44a4e7/STM32F1/libraries/USBComposite/usb_setup.cpp#L48-L53

rogerclarkmelbourne, 2019-02-18:
BTW. This is not a bug. So I'm now closing this issue.

closed 2019-02-18 by rogerclarkmelbourne

l15y, 2019-02-18:
Thanks.
Infact,I have tried your code,and have the same phenomenon as mine

stevstrong, 2019-02-18:
On a Windows PC the re-enumeration can take up to 5 seconds.

tinwhisker, 2019-03-01:
FYI, my cheapy "blue pill" doesn't react on twiddling PA12 directly.

Instead I bridged PA15 to PA12 and use similar code to the above (output low, then input floating).
I had consistent results with 100ms, where 50ms didn't work at all.
Even 500ms was pretty stable.

Tested on Windows 10 and macOS.

rogerclarkmelbourne, 2019-03-01:
@tinwhisker 

Interesting. 

Its been suggested that the complex reset circuit used on by the Maple Mini was not really needed, since the GPIO pins default to INPUT (high impedance) so would not cause a problem on the USB if they were bridged.

Did you check what value pullup resistor is on your board. The usual problem is often that the board has a 10k pullup resistor, but it should have 1.5k

tinwhisker, 2019-03-01:
@rogerclarkmelbourne

Yes, I switched out the 10k for a 1.5k, the board otherwise works, enumerating on all OS's I've stuck it on.

I got led to the workaround as the board was booting as 'maple' and not the HID device I set on it. I wasn't seeing a real benefit to the tweaks using a transistor, as it still used an IO pin, and it was only pulling down - extra components for what?! (I suppose if a HW fix was the only way, an RC reset might be a tolerable bodge).
Initially thinking it was not storing the sketch, I found your comment about re-enumeration, though trying the PA12 trick direct wasn't making a difference.
Even setting it low indefinitely, the host was still seeing the initial device, where forcing a pull down would drop the device.

Admittedly, I only gave it brief attention before bridging to PA15 and calling it a day when it worked.
I'll revisit it monday, as it lives on my work desk now - after all, the HID library I'm using could very well be resetting the port, despite its .begin() not called yet!

rogerclarkmelbourne, 2019-03-02:
@tinwhisker 

I'm not sure what the HID library does ;-)

Even the hack of setting PA12 to GPIO to set it LOW, was never really given much investigation.
@victorpv was the first person to try it, and it worked for him, and it also works for most people.
But there are some hardware configs e.g. Mac's with keyboard hubs where its problematic.

Also.. The latest RaspberryPi 3B+ also has a problem with enumeration, and it requires a longer pulse.