rogerclarkmelbourne/Arduino_STM32 #600: Issue when uploading on MacOS with path containing white space / blank

gerardwr, 2019-02-19:
**Describe the bug**
Upload to STM32 (maple_upload) fails when stm32duino is installed in a directory with a name containing a blank / white space.
 
**To Reproduce**

STM32duino installed in directory (space in "My Data"):
```
/Users/gerard/Documents/My Data/ArduinoESP/hardware/Arduino_STM32-master
```
Upload results in (sorry, is in Dutch):
```
De schets gebruikt 16132 bytes (12%)  programma-opslagruimte. Maximum is 131072 bytes.
Globale variabelen gebruiken 3088 bytes (15%) van het dynamisch geheugen. Resteren 17392 bytes voor lokale variabelen. Maximum is 20480 bytes.
/Users/gerard/Documents/My Data/ArduinoESP/hardware/Arduino_STM32-master/tools/macosx/maple_upload cu.usbmodemFA411 2 1EAF:0003 /var/folders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_build_316844/HelloWorld_STM32.ino.bin 
/Users/gerard/Documents/My Data/ArduinoESP/hardware/Arduino_STM32-master/tools/macosx/maple_upload: line 34: /Users/gerard/Documents/My: No such file or directory
Er trad een error op gedurende laden van de schets
```

** Workaround
As a quick fix I adapted the maple_upload source:

* Line 34 : replaced `$(dir)` with `"${DIR}"`
* Lines 44,48,53 : replaced `${DFU_UTIL}` with `"${DFU_UTIL}"`

I'm sure there's a neater way to fix the issue for all users.

stevstrong, 2019-02-19:
Why don't you issue a PR?

gerardwr, 2019-02-19:
@stevstrong 
I would have been happy to issue a PR.

I'm not an expert on Github so opening an "issue" is the only way I know to report a "problem".

Out of curiousity : How DO you report a PR?

stevstrong, 2019-02-19:
G**gle is your friend: https://www.google.com/search?client=firefox-b-d&q=github+issue+a+pull+request

gerardwr, 2019-02-19:
@stevstrong 
I get it now, in my world a PR is a Problem Report, you meant a Pull Request.

stevstrong, 2019-09-15:
This is obviously solved by https://github.com/rogerclarkmelbourne/STM32duino-bootloader/issues/61
If not, you can still reopen it.

closed 2019-09-15 by stevstrong