rogerclarkmelbourne/Arduino_STM32 #605: Support for low power modes

TamojitSaha, 2019-03-07:
Is there any support for low power modes with this core?
Like [here](https://github.com/stm32duino/STM32LowPower)

stevstrong, 2019-09-16:
I don't think this will ever find a way in our repo here.
Unless some has time to do that.

rogerclarkmelbourne, 2019-09-16:
I think its up to @TamojitSaha to submit a PR for the changes. 

This is not a commercial project where you can request that work be done. Its a community project and its up to each member of the community to do some of the work and submit their work to the community so that every can benefit.

I'd close the issue and the OP can re-open it or submit a PR when they have ported the code.

rogerclarkmelbourne, 2020-05-18:
Closed because of no activity

closed 2020-05-18 by rogerclarkmelbourne