rogerclarkmelbourne/Arduino_STM32 #606: EEPROM multiple data at the same address

Dave4675, 2019-03-12:
I have a bluepill of 128KB with a bootloader loaded, I really apreciate this repo, so thank you for that.

I'm tring to store some settings data into the EEPROM but when I load the example I found an issue (or maybe not) if I press 3 for what I know I expect to write at the location at 0x10, until I press 4 of course,, but when I press 5 it takes the next location, keeping the previous stored data that also become inaccessible.

If this issue is well known I wish to understand if I have to manage with the page address directly, and how to do that, otherwise I need to save all data into the ram, format the EEprom and write my new data again, wich I hope not.

Dave4675, 2019-03-13:
Okay, so I made some test, I tried to modify the EE_VerifyPageFullWriteVariable function, the one that calls the real function that writes the half word into the EEprom, or flash to be more precise. 
i made some changes that give the result that I want, but is still not possible to write to the same virtual address till two conditions, or the free side is equal to 0xFFFF wich means that the flash or the page has been erased or the data that we want to write is less than the one already written.
Also the update function is useless because this function already verify this.

this is my modified function:

uint16 EEPROMClass::EE_VerifyPageFullWriteVariable(uint16 Address, uint16 Data)
{
	FLASH_Status FlashStatus;
	uint32 idx, pageBase, pageEnd, newPage;
	uint16 count;
	boolean found = false;
	// Get valid Page for write operation
	
	pageBase = EE_FindValidPage();
	if (pageBase == 0)
		return  EEPROM_NO_VALID_PAGE;
	
	// Get the valid Page end Address
	pageEnd = pageBase + PageSize;			// Set end of page
	
	for (idx = pageEnd - 2; idx >= pageBase; idx -= 4)
	{
		if ((*(__IO uint16*)idx) == Address)		// Find last value for address
		{
			found = false;
			
			count = (*(__IO uint16*)(idx - 2));	// Read last data
			
			if (count == Data)
				return EEPROM_OK;
			
			if (count == 0xFFFF || Data < count)
			{
			
			FlashStatus = FLASH_ProgramHalfWord(idx - 2, Data);	// Set variable data
			if (FlashStatus == FLASH_COMPLETE)
				return EEPROM_OK;
			}
			
			break;
		} else {
			found = true;
			
		}
	}
	if(found){
				// Check each active page address starting from begining
				for (idx = pageBase + 4; idx < pageEnd; idx += 4){
					if ((*(__IO uint32*)idx) == 0xFFFFFFFF)				// Verify if element 
				{													//  contents are 0xFFFFFFFF
					FlashStatus = FLASH_ProgramHalfWord(idx, Data);	// Set variable data
					if (FlashStatus != FLASH_COMPLETE)
					return FlashStatus;
					FlashStatus = FLASH_ProgramHalfWord(idx + 2, Address);	// Set variable virtual address
					if (FlashStatus != FLASH_COMPLETE)
						return FlashStatus;
					return EEPROM_OK;
				}
				}
	}
	
	// Empty slot not found, need page transfer
	// Calculate unique variables in page
	count = EE_GetVariablesCount(pageBase, Address) + 1;
	if (count >= (PageSize / 4 - 1))
	
		return EEPROM_OUT_SIZE;
		
	if (pageBase == PageBase1)
		newPage = PageBase0;		// New page address where variable will be moved to
	else
		newPage = PageBase1;
	
	// Set the new Page status to RECEIVE_DATA status
	FlashStatus = FLASH_ProgramHalfWord(newPage, EEPROM_RECEIVE_DATA);
	if (FlashStatus != FLASH_COMPLETE)
		return FlashStatus;

	// Write the variable passed as parameter in the new active page
	FlashStatus = FLASH_ProgramHalfWord(newPage + 4, Data);
	if (FlashStatus != FLASH_COMPLETE)
		return FlashStatus;

	FlashStatus = FLASH_ProgramHalfWord(newPage + 6, Address);
	if (FlashStatus != FLASH_COMPLETE)
		return FlashStatus;

	return EE_PageTransfer(newPage, pageBase, Address);
	
}

EEPROMClass::EEPROMClass(void)
{
	PageBase0 = EEPROM_PAGE0_BASE;
	PageBase1 = EEPROM_PAGE1_BASE;
	PageSize = EEPROM_PAGE_SIZE;
	Status = EEPROM_NOT_INIT;
}

The result is that when we want to try to write to the same location there is an erase of the page0 and all data are passed to the page1 and viceversa if we try again. 
Said that this means that we have only one page that we can use, but whit the old function I tried to fulfill the page and this function does not provide to write to the other page it just returns the eeprom full error.
Also the real problem with the old function is that by writing multiple data at the same address we lose space, fulfilling the page, at that point we are forced to read the eeprom store all data into the ram, erase the page and then we can write everything from the ram to the page, in this way we gain the space we loosed before.

With the modified function we can use an entire page without managing the fulfilled data at the end but of course every time we want to write to the same address we are ruining the flash.

Probably we can modify the function so that when the page is full we can use the ram instead of the other page, in this way we ruin only one page at once.
Also we need to manage the exception of page fulfilled by starting to write to the page1.

I don't know if this isses were well known, but this flash with the __IO command works like that

rogerclarkmelbourne, 2019-03-13:
Thanks

I don't have time to look at this at the moment.

In general the EEPROM lib does the bare minimum to emulate the Arduino AVR EEPROM library, but probably needs to be rewritten.

However I doubt anyone has the time do that.

You may also want to consider using ST's own official Arduino core, (but I don't know for sure whether that core has a EEPROM library either)

fpistm, 2019-03-13:
> You may also want to consider using ST's own official Arduino core, (but I don't know for sure whether that core has a EEPROM library either)

EEPROM emulation is available.

Dave4675, 2019-03-13:
I know that there is the "official core" but this version works better with some library that I use, and also the other library doesn't support the loading through the usb port, I tried with it, probably because I didn't take enough time to understand.

rogerclarkmelbourne, 2019-03-13:
@fpistm 

Thanks

stevstrong, 2019-09-15:
@Dave4675 
could you please tell us how did you solved your problem? Others may be also interested in the solution.

Dave4675, 2019-09-19:
I have written the function that you need to change, mine has changes applied, you can find it at the top of this discussion

victorpv, 2019-12-27:
I'm not sure this is an issue. I couldn't understand the issue from the description above.

Rewriting single words in flash is possible. To rewrite you need to first erase the whole page. Since a flash page is estimated to last about 1000 erase cycles, clearing the page each time a single word changes would wear out the flash extremely fast. For that reason it uses the following technique:
-It uses 2 pages of flash. One is active and one is spare.
-It splits each 32bit word in flash in 2 chunks, 16bit for an address number, and 16bit for data.
-When the user tries to write a value to an address (i.e write 256 to address 1), the library will scan the active page for any word with 1 as the address half. If it finds it, reads the data. If the data is 256, then writes nothing since it already has 256 for address 1. If the function finds an word with address 1 but different data, or doesn't fine any word with address 1, it will write the next free word in the active flash page with that content (32bits word, address field value 1, data field value 256).
-When reading data, it will start the search from the end of the active page. So only the most current word with the matching address field is read. Example, if I want to read address 1, it will start from the end of the active page. Once it finds a word with address field = 1, it stops looking.
-If I write address 1 with value 256, and next I write it with value 254, it will NOT delete the previous word with address 1, it will just write the address and new data in the first empty word in the active page. Since the search for reads start from the end of the page, it will find the last value I ever wrote that matches the address field.
-If the current active page happens to be full, it will read the content of that page starting from the end and transfer it to the spare page. It will only write the last value for any address (so I wrote address 1 three times in the previous page, it will only copy the very latest value and skip the others, thus reducing the number of words actually copied to the spare page). Once the copy is complete, it marks the spare page as active, and clears the previously active page and marks it as spare.

This whole mechanism doesn't make a good usage of the capacity of 2 pages, but it considerably reduces the number of erase cycles need to rewrite the same addresses over and over, which a common occurrence.
The library is based on an official STM application note, I advise to read the application note for a better explanation.

Given all the above and what I have seen posted in this issue, I believe the library is working as it should.
If not, could you try to describe the issue in a different way, perhaps I can understand it better in a different way.

stevstrong, 2020-12-19:
I also had a closer look, the original post does not seem to be an issue, but the way how it should work, see the link provided by @victorpv .
So I will close this.

closed 2020-12-19 by stevstrong