rogerclarkmelbourne/Arduino_STM32 #608: Not compiling examples on STM32ADC

fa1ke5, 2019-03-22:
SingleChannelSingleConversion:8: error: 'ADC' does not name a type
 ADC myAdc(ADC1);
 ^
/home/madmax/Arduino/hardware/Arduino_STM32/STM32F1/libraries/STM32ADC/examples/SingleChannelSingleConversion/SingleChannelSingleConversion.ino: In function 'void int_func()':
SingleChannelSingleConversion:13: error: 'adc_result' was not declared in this scope
   Serial.println(adc_result);
                  ^
/home/madmax/Arduino/hardware/Arduino_STM32/STM32F1/libraries/STM32ADC/examples/SingleChannelSingleConversion/SingleChannelSingleConversion.ino: In function 'void setup()':
SingleChannelSingleConversion:18: error: 'myADC' was not declared in this scope
   myADC.setTrigger(ADC_EXT_EV_SWSTART);//start on SWStart bit
   ^
exit status 1
'ADC' does not name a type

**********************************************

what am I doing wrong?

stevstrong, 2019-03-22:
Check out #610 and let us know if it works.

fa1ke5, 2019-03-22:
now i have

/tmp/arduino_modified_sketch_718014/SingleChannelSingleConversion.ino: In function 'void int_func()':
SingleChannelSingleConversion:13: error: 'adc_result' was not declared in this scope
   Serial.println(adc_result);
                  ^
/tmp/arduino_modified_sketch_718014/SingleChannelSingleConversion.ino: In function 'void setup()':
SingleChannelSingleConversion:18: error: 'myADC' was not declared in this scope
   myADC.setTrigger(ADC_EXT_EV_SWSTART);//start on SWStart bit
   ^
exit status 1
'adc_result' was not declared in this scope

stevstrong, 2019-03-22:
OK, I fixed those errors, try it again.
You may eventually need to adapt the analog input pin.

fa1ke5, 2019-03-23:
D:\users\madmax\Documents\Arduino\sketch_mar23a\sketch_mar23a.ino: In function 'void setup()':

sketch_mar23a:22:9: error: 'class STM32ADC' has no member named 'attachADCInterrupt'

   myADC.attachADCInterrupt(int_func);

         ^

exit status 1
'class STM32ADC' has no member named 'attachADCInterrupt'

stevstrong, 2019-03-24:
It compiles now for me, check it yourself.

fa1ke5, 2019-03-24:
May are you publishing full final code?

stevstrong, 2019-03-24:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/1167371b13303fe7ad293fcc7f480f608755261b/STM32F1/libraries/STM32ADC/examples/SingleChannelSingleConversion/SingleChannelSingleConversion.ino

fa1ke5, 2019-03-24:
Nice work! Its comiling now, but dont working.
No output in serial....
While compiling i can see this error, but code succesly loading in blue pill .

libraries\STM32ADC\STM32ADC.cpp.o: In function `STM32ADC::setChannels(unsigned char*, unsigned char)':

D:\Users\madmax\Documents\Arduino\hardware\Arduino_STM32\STM32F1\libraries\STM32ADC\src/STM32ADC.cpp:113: warning: undefined reference to `adc_set_reg_seq_channel(adc_dev*, unsigned char*, unsigned char)'

fa1ke5, 2019-03-24:
of corse im changing GPIO pin before compilation

fa1ke5, 2019-03-24:
//myADC.setChannels(pin, 1); //this is actually the pin you want to measure
myADC.setPins(pin, 1); //this is actually the pin you want to measure

now it working!
Thanks, and nice day!

closed 2019-03-26 by fa1ke5