rogerclarkmelbourne/Arduino_STM32 #613: problem with IIC2

poojaasritha, 2019-03-25:
error while uncommenting the below line in i2c scanner example....
please fix it asap.................
and also show us a way on how to use I2C2 LINE with Wire_slave.h
//TwoWire WIRE2 (2,I2C_FAST_MODE);

error that i'm getting is :
'WIRE2' was not declared in this scope

DeadSHoT555, 2019-03-25:
I also face same problem.
Please show us a quick solution to use that

rajjjjjh, 2019-03-25:
Yeah i too can't communicate with second  i2c line...  How to communicate on second i2c line??????

GitMoDu, 2019-03-26:
`#include <Wire_slave.h> `

Replace with 

`#include <Wire.h> `

Now you can instantiate your prefered I2C channel.

```
TwoWire WireInstance(2,I2C_FAST_MODE);

WireInstance.beginTransmission(adderss)
```

rikykumar, 2019-04-02:
in that case the error was
  Wire.onRequest and Wire.onRecieve were not declared

rikykumar, 2019-04-02:
could not use stm32f103c as slave with <Wire.h>

rikykumar, 2019-04-02:
do u have any option to use stm32 as slave on Wire2

GitMoDu, 2019-04-02:
I2C slave is not supported on STM32 duino, on either channel.

rikykumar, 2019-04-02:
no.., using <Wire_slave.h> it is possible to communicate as slave with stm32 ..., i have tested it .., but i dont know how to use ion on second wire

rikykumar, 2019-04-02:
there are direct inbuild examples with package to user stm32 as slave under "Wire_slave" folder

LMESTM, 2019-04-02:
I think you can use other instances of I2C (there can be more than 2) by using an alternative, declaration with SCL and SDA pins:     TwoWire(uint8_t sda, uint8_t scl);

Like:
TwoWire Wire2(PB_11, PB_10);

Wouldn't that unblock the situation ?

GitMoDu, 2019-04-02:
You're correct, it compiles at least. As far as I knew, I2C support wasn't there for slave. I don't how much more I can help you.

DeadSHoT555, 2019-04-02:
[image]
Couldn't get it

DeadSHoT555, 2019-04-02:
[image]
And while using <Wire.h> could'nt work out slave mode but can hanle Wire2

stevstrong, 2019-09-15:
Possible duplicate of #637, #638 .

stevstrong, 2019-09-15:
Needs implementing an example to simultaneously use Wire as master and slave.
Possible duplicate of #625

PR #679 (2019-10-04): WireSlave Library Rework to make I2C Slave Fully Functional
Rework the WireSlave library of the STM32F1 target, merging it with the core, and fixing all of the I2C quirks and problems.

This fixes #625 and fixes #613 and fixes #207 and possibly other open issues.

I have tested the new WireSlave code and I2C core enhancements as both a master, communicating with CAT24C256 EEPROMs and MCP4725 DACs, and as a slave, communicating with a STM32F746G Discovery board and with the [I2CDriver](https://i2cdriver.com/) test gadget (a very handy tool, BTW).

The WireSlave examples have also been updated w [... cut]
closed 2019-10-04 by stevstrong