rogerclarkmelbourne/Arduino_STM32 #618: STM32F103 CAN library issue

akhilasrees, 2019-04-01:
I downloaded the CAN library for STM32F103 from this link: https://github.com/Phonog/Arduino_STM32

When I compile the Hardware CAN example code, I am getting these errors.

Arduino: 1.6.5 (Windows 8.1), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STLink, 72Mhz (Normal), Smallest (default)"

Build options changed, rebuilding all

Using library HardwareCAN in folder: C:\Users\hp\Documents\Arduino\libraries\HardwareCAN 

C:\Users\hp\AppData\Roaming\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10605 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1 -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c -IC:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src C:\Users\hp\AppData\Local\Temp\build7020112209124217676.tmp\HardwareCANexample.cpp -o C:\Users\hp\AppData\Local\Temp\build7020112209124217676.tmp\HardwareCANexample.cpp.o 

C:\Users\hp\AppData\Roaming\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10605 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1 -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c -IC:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\HardwareCAN.cpp -o C:\Users\hp\AppData\Local\Temp\build7020112209124217676.tmp\HardwareCAN\HardwareCAN.cpp.o 

C:\Users\hp\AppData\Roaming\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc -c -g -Os -DDEBUG_LEVEL=DEBUG_NONE -std=gnu11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10605 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1 -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple -IF:\Arduino\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c -IC:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c -o C:\Users\hp\AppData\Local\Temp\build7020112209124217676.tmp\HardwareCAN\utility\can.c.o 

C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c: In function 'can_deinit':
C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c:151:17: error: 'RCC_CAN' undeclared (first use in this function)
   rcc_reset_dev(RCC_CAN);
                 ^
C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c:151:17: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c: In function 'can_init':
C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c:179:17: error: 'RCC_CAN' undeclared (first use in this function)
  rcc_clk_enable(RCC_CAN);   // and CAN
                 ^
C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c: At top level:
C:\Users\hp\Documents\Arduino\libraries\HardwareCAN\src\utility\can.c:280:152: error: expected ';', ',' or ')' before '=' token
 CAN_STATUS can_filter(CAN_Port* CANx, uint8 filter_idx, CAN_FIFO fifo, CAN_FILTER_SCALE scale, CAN_FILTER_MODE mode, uint32 fr1, uint32 fr2, int extID = 0)
                                                                                                                                                        ^
Error compiling.

I dont understand what to do. Please help.

GitMoDu, 2019-04-02:
I don't understand. You downloaded a fork, but came to request support on the original repo?

akhilasrees, 2019-04-02:
Ok I am sorry. Can you please give me the link to the actual library for CAN in STM32F103C8T6

GitMoDu, 2019-04-02:
If you're using this fork: https://github.com/Phonog/Arduino_STM32

You should use their libraries and issues.

akhilasrees, 2019-04-02:
If I want to work with actual libraries for CAN in STM32F103 using MCP2551, where should I search for it?

stevstrong, 2019-09-15:
I think this comment answers your question: https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/618#issuecomment-478947924.

closed 2019-09-15 by stevstrong