rogerclarkmelbourne/Arduino_STM32 #619: Location of platformio-build.py file??

SJ-Innovation, 2019-04-04:
Hi,  Thanks for your awesome work on this framework by the way. 

Im looking at making a couple of minor changes to platformio-build.py (tools->platformio-build.py) which is available from the CDN packaged versions of the framework that platformio downloads and uses ([Here](https://bintray.com/platformio/dl-packages/framework-arduinoststm32)).

However, i cant for the life of me find where this file is developed! I have searched high and low for where this damn file comes from but i'm at a bit of a wits end... I dont suppose you could lend an insight could you? 

Thanks, and all the best... 
Sam

fpistm, 2019-04-04:
Hi  @SJ-Innovation 
This script is not in this repo. It is one managed by platformio.
You should ask here:
https://github.com/platformio/platform-ststm32

Probably you search this one:
https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino.py

SJ-Innovation, 2019-04-04:
@fpistm  I have been searching there, was just about to ask there before you replied!  Regrettably though that repo is of the 'platform' files, whereas the script im looking to edit is part of the 'package' files. I can only assume that you're correct in that they are added by the pio guys. Ill contact them! Thank you for your time! 

- Sam

closed 2019-04-04 by SJ-Innovation