rogerclarkmelbourne/Arduino_STM32 #620: "sprintf" and "dtostrf" does not work when compiling as "--specs=nano.specs" added in platform.txt

HamidSaffari, 2019-04-04:
Hi and best regards to creators.
"sprintf" and "dtostrf" does not work properly when compiling as "--specs=nano.specs" option added to the platform.txt .
as discussed here:
 http://www.stm32duino.com/viewtopic.php?f=28&t=1596&p=20642#p20627 
by adding :   --specs=nano.specs   at the very end of line in Documents\Arduino\hardware\Arduino_STM32\STM32F1\platform.txt on the line that starts with "recipe.c.combine.pattern=...."
 the flash usage reduced to almost half, but in price of loosing some huge-flash-using functions as they said. and one of them that I realized is "sprintf" and "dtostrf" in some form of use like in my case floats: sprintf(buff, "%5.1f", Temperature_C);
but since the Arduino_Core_STM32:  https://github.com/stm32duino/Arduino_Core_STM32 
has used that option by default, he also has written new version of "dtostrf" that works perfectly with this option and I used their function in Arduino_STM32 platform and works fine. here it is:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/avr/dtostrf.c
now I think it's better replace this with your dtostrf.c library.
and also using "--specs=nano.specs" option by default and see the other lost to try fix it.

stevstrong, 2019-09-15:
What is the status of this issue?

stevstrong, 2019-09-16:
@HamidSaffari 
Can you please provide a PR?

HamidSaffari, 2019-09-22:
when you use "sprintf(buff, "%5.1f", Temperature_C);" with original framework and choose
"--specs=nano.specs" for compile the output is empty, but when U do what I described then it print out the number correctly.

danieltr3s, 2020-05-25:
@stevstrong I think this [pull request](https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/790) is what @HamidSaffari was requesting.

Also see this [Pull Request](https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/787)

Best regards,
Tr3s

stevstrong, 2020-10-12:
I will check the first PR.

stevstrong, 2021-01-24:
#790 is a fix for this and was merged so that this can be closed.

closed 2021-01-24 by stevstrong