rogerclarkmelbourne/Arduino_STM32 #621: How to program blue STM32F407VET with st-link v2 and stm32duino?

drelephant, 2019-04-05:
I'm using Arduino 1.8.5 on Windows 7 x64. I installed stm32duino by following [these instructions from the stm32duino wiki](https://wiki.stm32duino.com/index.php?title=Installation#Arduino_and_Arduino_STM32). It installed the latest Arduino SAM Boards version 1.6.12.

I installed into "C:\Users\username\Documents\Arduino\hardware\Arduino_STM32" (I removed the "-master" from the end).

I can connect to the STM32F407VET6 using the ST-LINK Utility, so it must be connected OK.
[image]

I can successfully use my ST-Link V2 to flash a stm32f103c blue pill, but with my [Vcc-gnd.com STM32F407VET6 Mini](https://wiki.stm32duino.com/index.php?title=Vcc-gnd.com_STM32F407VET6_Mini) it always fails.

Here are the upload setting I've chosen:

[image]

Here is the compilation/upload output:
`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\username\AppData\Local\Arduino15\packages -hardware C:\Users\username\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\username\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\username\Documents\Arduino\libraries -fqbn=Arduino_STM32:STM32F4:generic_f407v:usb_cfg=usb_nc,upload_method=STLinkMethod,opt=osstd -ide-version=10805 -build-path C:\Users\username\AppData\Local\Temp\arduino_build_995520 -warnings=none -build-cache C:\Users\username\AppData\Local\Temp\arduino_cache_19761 -prefs=build.warn_data_percentage=75 -verbose C:\Users\username\Documents\Arduino\DoubleBlink_STM32F407VET_blue_board\DoubleBlink_STM32F407VET_blue_board.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\username\AppData\Local\Arduino15\packages -hardware C:\Users\username\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\username\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\username\Documents\Arduino\libraries -fqbn=Arduino_STM32:STM32F4:generic_f407v:usb_cfg=usb_nc,upload_method=STLinkMethod,opt=osstd -ide-version=10805 -build-path C:\Users\username\AppData\Local\Temp\arduino_build_995520 -warnings=none -build-cache C:\Users\username\AppData\Local\Temp\arduino_cache_19761 -prefs=build.warn_data_percentage=75 -verbose C:\Users\username\Documents\Arduino\DoubleBlink_STM32F407VET_blue_board\DoubleBlink_STM32F407VET_blue_board.ino
Using board 'generic_f407v' from platform in folder: C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4
Using core 'maple' from platform in folder: C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4
Detecting libraries used...
"C:\Users\username\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall  -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_f407v -DVECT_TAB_FLASH -DUSER_ADDR_ROM=(uint32)0x08000000 -DERROR_LED_PORT=GPIOA -DERROR_LED_PIN=7 -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=168000000L -DARDUINO=10805 -DARDUINO_STM32GenericF407VET6 -DARDUINO_ARCH_STM32F4  -mthumb -DSTM32_HIGH_DENSITY -DSTM32F4 -DBOARD_generic_f407v -DUSB_NC -DVECT_TAB_FLASH -DUSER_ADDR_ROM=(uint32)0x08000000  -mthumb -D__STM32F4__ "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\system/libmaple" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\variants\generic_f407v" "C:\Users\username\AppData\Local\Temp\arduino_build_995520\sketch\DoubleBlink_STM32F407VET_blue_board.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\username\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall  -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_f407v -DVECT_TAB_FLASH -DUSER_ADDR_ROM=(uint32)0x08000000 -DERROR_LED_PORT=GPIOA -DERROR_LED_PIN=7 -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=168000000L -DARDUINO=10805 -DARDUINO_STM32GenericF407VET6 -DARDUINO_ARCH_STM32F4  -mthumb -DSTM32_HIGH_DENSITY -DSTM32F4 -DBOARD_generic_f407v -DUSB_NC -DVECT_TAB_FLASH -DUSER_ADDR_ROM=(uint32)0x08000000  -mthumb -D__STM32F4__ "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\system/libmaple" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\variants\generic_f407v" "C:\Users\username\AppData\Local\Temp\arduino_build_995520\sketch\DoubleBlink_STM32F407VET_blue_board.ino.cpp" -o "C:\Users\username\AppData\Local\Temp\arduino_build_995520\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\username\AppData\Local\Temp\arduino_build_995520\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\username\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -MMD -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_f407v -DVECT_TAB_FLASH -DUSER_ADDR_ROM=(uint32)0x08000000 -DERROR_LED_PORT=GPIOA -DERROR_LED_PIN=7 -mcpu=cortex-m4 -DF_CPU=168000000L -DARDUINO=10805 -DARDUINO_STM32GenericF407VET6 -DARDUINO_ARCH_STM32F4  -mthumb -DSTM32_HIGH_DENSITY -DSTM32F4 -DBOARD_generic_f407v -DUSB_NC -DVECT_TAB_FLASH -DUSER_ADDR_ROM=(uint32)0x08000000  -mthumb -D__STM32F4__ "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\system/libmaple" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple" "-IC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\variants\generic_f407v" "C:\Users\username\AppData\Local\Temp\arduino_build_995520\sketch\DoubleBlink_STM32F407VET_blue_board.ino.cpp" -o "C:\Users\username\AppData\Local\Temp\arduino_build_995520\sketch\DoubleBlink_STM32F407VET_blue_board.ino.cpp.o"
Compiling libraries...
Compiling core...
Using previously compiled file: C:\Users\username\AppData\Local\Temp\arduino_build_995520\core\stm32_isrs.S.o
Using previously compiled file: C:\Users\username\AppData\Local\Temp\arduino_build_995520\core\stm32_vector_table.S.o
Using previously compiled file: C:\Users\username\AppData\Local\Temp\arduino_build_995520\core\wirish\start.S.o
Using previously compiled file: C:\Users\username\AppData\Local\Temp\arduino_build_995520\core\pin_map.c.o
Using previously compiled file: C:\Users\username\AppData\Local\Temp\arduino_build_995520\core\wirish\start_c.c.o
Using previously compiled file: C:\Users\username\AppData\Local\Temp\arduino_build_995520\core\generic_f407v.cpp.o
Using precompiled core
Linking everything together...
"C:\Users\username\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -Os -Wl,--gc-sections  -mcpu=cortex-m4 "-TC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\variants\generic_f407v/ld/jtag.ld" "-Wl,-Map,C:\Users\username\AppData\Local\Temp\arduino_build_995520/DoubleBlink_STM32F407VET_blue_board.ino.map" "-LC:\Users\username\Documents\Arduino\hardware\Arduino_STM32\STM32F4\variants\generic_f407v/ld" -o "C:\Users\username\AppData\Local\Temp\arduino_build_995520/DoubleBlink_STM32F407VET_blue_board.ino.elf" "-LC:\Users\username\AppData\Local\Temp\arduino_build_995520" -lm -lgcc -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "C:\Users\username\AppData\Local\Temp\arduino_build_995520\sketch\DoubleBlink [... cut ...]
`C:\Users\username\Documents\Arduino\hardware\Arduino_STM32/tools/win/stlink_upload.bat COM55 {upload.altID} {upload.usbID} C:\Users\username\AppData\Local\Temp\arduino_build_995520/DoubleBlink_STM32F407VET_blue_board.ino.bin`

Whereas when I do the upload for the stm32f103c blue pill, the command line is:
`C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>stlink\ST-LINK_CLI.exe -c SWD -P C:\Users\username\AppData\Local\Temp\arduino_build_995520\DoubleBlink_STM32F407VET_blue_board.ino.bin 0x8000000 -Rst -Run -NoPrompt`

How can I fix this?

stevstrong, 2019-04-05:
Please replace the line in: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/platform.txt#L127
with this one:
tools.stlink_upload.upload.pattern="{path}/{cmd}" "{build.path}/{build.project_name}.bin"

drelephant, 2019-04-05:
Thank you so much! It looks like it's programming now.

So... my new problem - the LED is not flashing...

I uploaded the following simple sketch - the led is supposedly on PB9:

```
int ledpin = PB9;

void setup() {
  pinMode(ledpin, OUTPUT);
}

void loop() {
  digitalWrite(ledpin, LOW);
  delay(1000);              // wait for a second
  digitalWrite(ledpin, HIGH);
  delay(50);
  digitalWrite(ledpin, LOW);
  delay(50);
  digitalWrite(ledpin, HIGH);
  delay(50);
}
```

Yet there's no LED flashing, even though the programming seems to have happened:

```
C:\Program Files (x86)\Arduino>rem @echo off 

C:\Program Files (x86)\Arduino>rem: Note C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win\ get path of this batch file 

C:\Program Files (x86)\Arduino>rem: Need to change drive if My Documents is on a drive other than C: 

C:\Program Files (x86)\Arduino>set driverLetter=C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win\ 

C:\Program Files (x86)\Arduino>set driverLetter=C: 

C:\Program Files (x86)\Arduino>C:

C:\Program Files (x86)\Arduino>cd C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win\ 

C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>rem: the two line below are needed to fix path issues with incorrect slashes before the bin file name 

C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>set str=C:\Users\username\AppData\Local\Temp\arduino_build_241339/DoubleBlink_STM32F407VET_blue_board.ino.bin 

C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>set str=C:\Users\username\AppData\Local\Temp\arduino_build_241339\DoubleBlink_STM32F407VET_blue_board.ino.bin 

C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>rem: ------------- use STLINK CLI 

C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>stlink\ST-LINK_CLI.exe -c SWD -P C:\Users\username\AppData\Local\Temp\arduino_build_241339\DoubleBlink_STM32F407VET_blue_board.ino.bin 0x8000000 -Rst -Run -NoPrompt 
STM32 ST-LINK CLI v3.3.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: Old ST-LINK firmware/ST-LINK already used
ST-LINK Firmware version: V2J17S4 (Need Update)
Connected via SWD.
Target voltage = 3.3 V
Connection mode: Normal
Reset mode: Software reset
Device ID: 0x413 
Device flash Size: 512 Kbytes
Device family: STM32F405xx/F407xx/F415xx/F417xx

Loading file...
Flash Programming:
  File : C:\Users\username\AppData\Local\Temp\arduino_build_241339\DoubleBlink_STM32F407VET_blue_board.ino.bin
  Address : 0x08000000
Memory programming...
±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±± 0%
 0%ÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛ 100%
Memory programmed in 1s and 544ms.
Programming Complete.

MCU Reset.

No ST-LINK detected

C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>rem: Using the open source texane-stlink instead of the proprietary STM stlink exe 

C:\Users\username\Documents\Arduino\hardware\Arduino_STM32\tools\win>rem:texane-stlink\st-flash.exe write C:\Users\username\AppData\Local\Temp\arduino_build_241339\DoubleBlink_STM32F407VET_blue_board.ino.bin 0x8000000 
```

The STM32F407VET was flashing when I first got it, but I did try to read the Flash with the ST-LINK Utility and then just write that .bin file back. It never flashed the LED again... And it seems I have to hold the reset button for a while when trying to connect to it or program it now, which I don't think I had to do before, although it does appear to program correctly.

Any ideas?

stevstrong, 2019-04-05:
The generic F4 mini board you have has an 25MHz quartz and I think that Roger's repo does not support  that.
As an alternative, you could try my repo which has good support for generic F4 boards:
https://github.com/stevstrong/Arduino_STM32
Don't forget to select the correct board type in Arduino IDE menu.
You can add issues to it if you find some.

drelephant, 2019-04-05:
Oh ok, that makes sense.

Odd that this board would be listed in the wiki as having "Good" Level of support - I bought it especially on the strength of that...

I'm starting to think this board is too difficult for me to use, I'm not an idiot but not an expert with this either.

I'm still having a problem even with your core, I will post an issue on your repo.

Thanks a lot for the help.

drelephant, 2019-04-05:
Actually I had the same problem with your core that [someone else reported](https://github.com/stevstrong/Arduino_STM32/issues/12). I never would have been able to do the fix without his explanation...

The LED is flashing now!! Thanks so much!!

drelephant, 2019-04-05:
Oops, closed the issue a bit early - I was still hoping for an answer on the best way to install both cores and choose between them?

stevstrong, 2019-04-05:
Don't know, try it.
I use only my libmaple core forked from Roger's repo.

rtek1000, 2019-05-09:
Hi, I would like to comment that sketch blink worked with the STM32-407 DIY-MORE board.

Thank you.
[image]
[image]

stevstrong, 2019-09-15:
In order to use Roger's core with the F4 development from my core you could try to replace the STM32F4 folder from Roger's core with the STM32F4 folder from my repo. It should work.
If does not, just reopen this issue.

closed 2019-09-15 by stevstrong