rogerclarkmelbourne/Arduino_STM32 #622: SPI::transfer(void *buf, size_t count) missing

7eggert, 2019-04-05:
Compiling the Ethernet card library fails due to a missing SPI method:

```
/opt/arduino-1.8.8/libraries/Ethernet/src/utility/w5100.cpp:315:22: error: no matching function for call to 'SPIClass::transfer(uint8_t [8], int)'
   SPI.transfer(cmd, 4);
                      ^
```

Would this implementation be correct? Should it go as an inline function into the header or into the cpp file?:

```
    inline void _transfer(void *buf, size_t count) {
      uint8 * buf8 = (uint8 *) buf;
      for (;count;count--,buf8++)
        *buf8 = transfer(*buf8);
    }
```

(OT: The Ethernet library seems to take 60 KB, while on Arduino, it takes 10 KB?)

stevstrong, 2019-04-05:
I don't know which library you are actually referring to, but you should better use the Ethernet_STM library from this repo. It is optimized for STM32.

7eggert, 2019-04-05:
I was talking about the default library from Arduino 1.6 or 1.8, `#include <Ethernet.h>`.

Thanks for recommending Ethernet_STM, it's only 41380 bytes now, about 10 KB for ethernet.

7eggert, 2019-04-08:
Also this is a long-standing issue and there already was a pull request (I only searched the issues, sorry):

github.com/rogerclarkmelbourne/Arduino_STM32/pull/586
or
https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/515
https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/516

This issue can be closed as soon as one of these is merged.

stevstrong, 2019-09-14:
I merged #586 .
So please test that and give feedback whether it works as expected or not.

7eggert, 2019-10-14:
/me is back in town. Did you merge that to a branch yet (as it's not in master)?

jjbubka, 2019-10-18:
Hi Steve, #586 is not merged. You have closed the PR without merging it. I have tested it with standard arduino ethernet (W5500) library and works well.
Thank you!

stevstrong, 2019-10-18:
Oh, I see my mistake.
But now I think I will rather consider #515 because it seems to be more optimized.
I just have to test it before merging it.

jjbubka, 2019-10-18:
Sounds good. 👍

stevstrong, 2020-12-13:
Solved by https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/73e264474ed3a3d4f4ee4c9497d394510509ab1f.

closed 2020-12-13 by stevstrong