rogerclarkmelbourne/Arduino_STM32 #625: can some one please suggest how to use I2C2 with <wire_slave> library

DeadSHoT555, 2019-04-14:
The example under i2c <Wire_slave.h>  for I2C2 Is giving compilation error..and including <SoftWire.h> along with <Wire_slave.h> is giving redeclaration error
could nt find how to use I2C2 with <Wire_slave.h>

stevstrong, 2019-09-11:
We would be very interested to see the code of a working version, so please share it if possible.

dewhisna, 2019-09-19:
Fixing this one certainly took more effort than I was expecting, no thanks the bazillion odd quirks that ST has in this I2C peripheral, but I am really pleased with how well it is performing now on my project where I replaced an underpowered ATmega328PB with a STM32F103CB.

And to @DeadSHoT555 who originally opened this issue, to readdress your original question, you can now simply use `#include <Wire_slave.h>` in your code.  Then use `Wire` for I2C1 and `Wire1` for I2C2.  There is no longer a cross-mangling of the `Wire.h` and `Wire_slave.h` headers.  The master-only `Wire` library uses `Wire.h` and the reworked `WireSlave` library (that works for both master and slave, but only for hardware I2C devices, not SoftWire) uses `Wire_slave.h`.  I personally compiled all of the examples under `WireSlave` after the rework and everything successfully compiles.

closed 2019-10-04 by stevstrong

rsonnicksen, 2023-05-24:
@dewhisna I'm trying to run a slave I2C interface on a "BluePill" (STM32F103C), and getting compile errors when using <Wire_slave.h> instead of <Wire.h>.  ArduinoIDE compiler is telling me TwoWire is protected.
Can you tell me how to fix?
Previously, I was declaring my object as:
`#include <Wire.h>

TwoWire HWire (1, I2C_FAST_MODE);`

In your post, I think you're implying that I2C1 and I2C2 are already defined in the library as "Wire" and "Wire1".
So do I simply start using these without declaring them?
Like this?
#include <Wire_slave.h>
Wire.begin(0x4C );
Wire.setClock(400000);

You mention **_merging the i2c changes from WireSlave back into the core_**
Can you provide more information, or provide files that I can replace?

Thanks!!

dewhisna, 2023-05-24:
> So do I simply start using these without declaring them?
Like this?
#include <Wire_slave.h>
Wire.begin(0x4C );
Wire.setClock(400000);

Yes, and you are correct that `Wire` and `Wire1` are already defined in the code.  The constructors for `TwoWire` are protected and so to use it, you have to  either use the existing class objects as defined or derive a new class from it.  Though your code for the calls above to `Wire` need to be inside of a function, such as the `setup` or `loop` function typically used in Arduino code.

You can also use a `#define` if you wanted to give it a different name to make your code more portable for compiling on other platforms and architectures by keeping the change to switch ports at one single place instead of having to change it throughout all of your code base.  In other words, if you used a `#define myWirePort Wire` and used `myWirePort` throughout the code, then if you ever wanted to change it to `Wire1`, you change that one line instead of all of the places in the code.

Also, the way the `WireSlave` code is designed with static instance reference functions, you could define your own variable even.   In your example above with `HWire`, you could do something like:

```
TwoWire& HWire = TwoWire::getInstance();
```

That would create a variable `HWire` that's a reference (essentially just a pointer) to the existing `Wire` object, but call it `HWire` instead and is more type-safe than using a `#define`  (note: use `getInstance1()` for `Wire1`).

As for my comment on merging `WireSlave` back into the core libraries, what I meant was that `WireSlave` was designed to be a replacement for `Wire`.  I didn't just flat-out replace it initially because others were using the existing `Wire` library as it was at the time and the API is slightly different.  `WireSlave` existed already to add slave device support, but had many issues.  And I needed it to work, so I rewrote `WireSlave` to fully function and do everything that `Wire` and `WireSlave` did but in one library.  The intent was for the owners of this repository to decide if/when at some point in the future they wanted to move `WireSlave` to `Wire` and replace `Wire` with it.  Doing so would just be a matter of moving the `WireSlave` source files to the `Wire` folder, renaming them to be `Wire` and fold the content of the `Wire.h` compatibility header with that of `Wire_slave.h`.

But the whole merging of it is only a technicality and the only advantages is that existing projects using `Wire` could just continue to include it and use it as `Wire` instead of having to use `WireSlave` for the include and library search paths (depending on your build environment).

PR #679 (2019-10-04): WireSlave Library Rework to make I2C Slave Fully Functional
Rework the WireSlave library of the STM32F1 target, merging it with the core, and fixing all of the I2C quirks and problems.

This fixes #625 and fixes #613 and fixes #207 and possibly other open issues.

I have tested the new WireSlave code and I2C core enhancements as both a master, communicating with CAT24C256 EEPROMs and MCP4725 DACs, and as a slave, communicating with a STM32F746G Discovery board and with the [I2CDriver](https://i2cdriver.com/) test gadget (a very handy tool, BTW).

The WireSlave examples have also been updated w [... cut]