rogerclarkmelbourne/Arduino_STM32 #626: Compile errors in the example sketch HardwareTimerPWMInput.ino

mrguen, 2019-04-26:
Hello,

Thanks for this great STM32 support repo. Looking for TIMER INPUT support I found your example sketch HardwareTimerPWMInput.ino but have some compile errors

1)
Streaming.h: No such file or directory
Then after removing this include:

2)
HardwareTimerPWMInput:28:9: error: no match for 'operator<<' (operand types are 'USBSerial' and 'const char [80]')

  Serial << "Example of the Timer Input Capture mode to measure PWM pulse width and period\n\n";

3)
HardwareTimerPWMInput:73:61: error: 'endl' was not declared in this scope

   Serial << ", period: " << Timer2.getCompare(TIMER_CH1) << endl;

stevstrong, 2019-04-26:
Yes, you are right.
It is about the ArduinoStreaming library available here: https://github.com/geneReeves/ArduinoStreaming/blob/master/Streaming.h
You can either copy the header file to your sketch folder or copy the whole library to /arduino/libraries folder.
One could eventually add `Streaming.h` to the repo.

mrguen, 2019-04-26:
Thanks. Anyway the standard Serial object was also working.

stevstrong, 2019-04-26:
Sureit works, but I like this kind of serial usage: 
`Serial << ", period: " << Timer2.getCompare(TIMER_CH1) << endl;`

mrguen, 2019-04-26:
Is there some reference documents for using these timers?

stevstrong, 2019-04-26:
http://docs.leaflabs.com/static.leaflabs.com/pub/leaflabs/maple-docs/latest/lang/api/hardwaretimer.html

stevstrong, 2019-04-26:
Trigger source for what?

stevstrong, 2019-04-26:
Check RM0008 section 14.3.19.

stevstrong, 2019-09-15:
I consider this as solved.
If it is not the case, just reopen it.

closed 2019-09-15 by stevstrong