rogerclarkmelbourne/Arduino_STM32 #627: Circular dependency in USB Composite

tmp64, 2019-04-26:
There is a circular dependency in USB Composite library. USBComposite.h includes USBMIDI.h, which includes USBComposite.h. Same thing with USBXBox360.h and USBMassStorage.h. That breaks Qt Creator's code model (shouldn't affect Arduino IDE though).

stevstrong, 2019-09-15:
I would expect that you provide a solution for this, as support for QT Creator is out of the scope of this repo.

tmp64, 2019-09-15:
`#include <USBComposite.h>` is not needed in these files. The documentation says that to use the library *USBComposite.h* needs to be included in the sketch. It defines `_USBCOMPOSITE_H_` as an include guard and then includes the files. Including *USBComposite.h* in them does nothing because `_USBCOMPOSITE_H_` was already defined so `#include <USBComposite.h>` is redundant.

stevstrong, 2019-09-15:
Can you please provide a PR?

tmp64, 2019-09-15:
#676

stevstrong, 2019-09-15:
Does this still compiles on Arduino IDE?

tmp64, 2019-09-15:
Yes, based on keyboardMouse, x360, midiOut and mass examples it does.

stevstrong, 2019-09-16:
I would like to double check whether it compiles error free.

rogerclarkmelbourne, 2020-05-18:
USB Composite is currently being reworked, and a PR is pending, which supersedes this issue

closed 2020-05-18 by rogerclarkmelbourne