rogerclarkmelbourne/Arduino_STM32 #628: [TouchScreen_STM] pressure calc error

rtek1000, 2019-04-26:
Hi,

Please check:

[image]

I tried to adapt another code for XPT2046, and found a possible error, note that pressing the impedance tends to decrease, and if we calculate wrong, the impedance tends to increase.

Not works here:
```
  float rTouch;
  rTouch = z2;
  rTouch /= z1;
  rTouch -= 1;
  rTouch *= x; // <------- (missing divide by 4095?)
  rTouch *= _RXPLATE;
```

Works here:
```
  float rTouch;
  rTouch = z2;
  rTouch /= z1;
  rTouch -= 1;
  rTouch *= float(x)/4095.0;
  rTouch *= _RXPLATE;
```

Thank you.

stevstrong, 2019-09-14:
Can you please provide a PR?

stevstrong, 2019-09-14:
Anyway, I think you missed the following lines of the code which do exactly what you are "fixing":
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/Touch-Screen-Library_STM/TouchScreen_STM.cpp#L188-L192

So the original code should work correctly if you use the Arduino framework where __STM32F1__ is defined at compile time.

stevstrong, 2019-09-15:
I think my previous comment provided the answer this so I am closing this.
You can still reopen it if you think this is still actual.

closed 2019-09-15 by stevstrong