rogerclarkmelbourne/Arduino_STM32 #629: Sketch HardTimerAsEncoder.ino runtime error

mrguen, 2019-04-27:
This instructon in the sketch HardTimerAsEncoder.ino generates an execution error. It seems the blue pill is in perpetual reset:   

timer.setMode(0, TIMER_ENCODER); //set mode, the channel is not used when in this mode. 

I could not find this TIMER_ENCODER mode in the timer library headers. I was looking for a way to count external incoming pulses for direct frequency counting. It seems HardwareTimer only implements internal clock based timers?

stevstrong, 2019-04-27:
There is no timer library.
The mentioned constant is defined here: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/9d46a1c27d4dc9dd4df06b76a1d81684519d8acc/STM32F1/system/libmaple/include/libmaple/timer.h#L573
You should consult the user's manual (RM0008) to find the appropriate timer mode to count pulses, the encoder mode is not designed for that.

mrguen, 2019-05-01:
Finally I could measure an signal on PA0 / ETR pin on timer 2 chained to timer 3 and gated by timer 1 (1 second gate period).

It works, but I can see many signal delays so the measurement is far from precise. I tried to gate timer 2 with:
 
* Timer 1 Enable signal MSM mode
* External signal generated by timer 1 PWM on pin A8 connected to pin A1 /TI2 gating
* Timer 1 OC1REF MSM mode

The less imprecise is the latest. Here is the code. If someone knows how to gate precisely the measured signal (ETR), that would be interesting. I am not interested in a reciprocal counter (given in STM32 Timers Cook Book),  since this is to measure high frequencies. I Tested at 4 MHz and got hundreds of hertz of error.

`
#include "Streaming.h"

int LSB = 0;
int MSB = 0;

void initCLK(void)
{
    // One PWM generated to test the counter
    // init timer for CLK generation on PA8 (timer 4 ch 1)
    pinMode(PB6, PWM);
    // generate 9MHz square signal
    pwmWrite(PB6, 1);
    // change here for different PWM frequency output on PB6
    timer_set_reload(TIMER4, 2);
}

void periodHandler() {

  (TIMER1->regs).gen->CR1 = 0; // Stop timer 1
  
  LSB = (TIMER2->regs).gen->CNT;
  MSB = (TIMER3->regs).gen->CNT;
  
  Serial << "f = " << ((MSB<<16) + LSB)  << endl; 
  
  (TIMER1->regs).gen->CNT = 0;
  (TIMER2->regs).gen->CNT = 0;
  (TIMER3->regs).gen->CNT = 0;

  (TIMER1->regs).gen->CR1 |= TIMER_CR1_CEN; // start time base
}

void TIMER_Setup(void)
{
    pinMode(PA0, INPUT);
    pinMode(PA1, INPUT);

    // TIME BASE ****************************
    timer_init(TIMER1);

    // See RM008 p. 399 "Using one timer to enable another timer"

    (TIMER1->regs).gen->CR2 = TIMER_CR2_MMS_COMPARE_OC1REF; // Master mode TRGO     
    (TIMER1->regs).gen->PSC = 7199;                         // Prescaler;
    (TIMER1->regs).gen->ARR = 10100;                        // Overflow 1.01 s
    (TIMER1->regs).gen->CCR1 = 10000;                       // Compare register = Measurement period 1s    
    (TIMER1->regs).gen->CCMR1 = (0b110U << 4);              // "110: PWM mode 1 - In upcounting, channel 1 is active as long as TIMx_CNT < TIMx_CCR1 else inactive." 
    (TIMER1->regs).gen->SMCR = TIMER_SMCR_MSM;              // Enable master/ slave mode 
    
    timer_attach_interrupt(TIMER1, 0,  periodHandler);   
 
    // EDGE COUNTER LSB ******************************
    timer_init(TIMER2);
    (TIMER2->regs).gen->CR1 = 0; // stop the timer
    (TIMER2->regs).gen->CR2 = 0;
    (TIMER2->regs).gen->CR2 = TIMER_CR2_MMS_UPDATE ; // Master mode TRGO on overflow RM008 p 399    
    (TIMER2->regs).gen->SMCR = 0;
    
    (TIMER2->regs).gen->SMCR = (TIMER_SMCR_ECE | TIMER_SMCR_SMS_GATED | TIMER_SMCR_TS_ITR0 | TIMER_SMCR_MSM); 
    
    (TIMER2->regs).gen->CCMR1 = TIMER_IC_INPUT_TRC ; // Works when TIMx_SMCR.TS is set. Here it is ITR0 master /slave signal from TIM1 to TIM2 signal 

    // EDGE COUNTER MSB **************************************
    timer_init(TIMER3);
    (TIMER3->regs).gen->CR1 = 0; // stop the timer
    (TIMER3->regs).gen->CR2 = 0;

    (TIMER3->regs).gen->SMCR = 0;
    (TIMER3->regs).gen->SMCR = (TIMER_SMCR_SMS | TIMER_SMCR_TS_ITR1); //Triggered by ITR1 TRGO from counter 2 see Table 86 RM0008 and page 399 chaining two timers as prescaler

    (TIMER3->regs).gen->DIER = 0;
    (TIMER3->regs).gen->SR = 0;
    (TIMER3->regs).gen->EGR = 0;

    (TIMER3->regs).gen->CCMR1 = 0; 

    (TIMER3->regs).gen->CCMR2 = 0;
    (TIMER3->regs).gen->CCER = 0;

    // Start timers
    (TIMER2->regs).gen->CR1 = TIMER_CR1_CEN; // start edge count LSB
    (TIMER3->regs).gen->CR1 = TIMER_CR1_CEN; // start edge count MSB    
    (TIMER1->regs).gen->CR1 = TIMER_CR1_CEN; // start time base

}

void setup()
{
  initCLK(); // init clock generation
  TIMER_Setup(); // set gated mode counting
  Serial.begin(250000);
  while(!Serial); delay(10);
}

void loop ()
{

}`

stevstrong, 2019-05-01:
I can imagine a simple way to measure pulses: input the pulses to a timer as clock and read the counter value periodically within an interrupt (it can be either generated by another timer or the systick).
In which range are the counts per second?

mrguen, 2019-05-02:
Yes apparently the interrupts fires before the MSM signal "totally" stops the slave counter.  If I add some delay in the interrupt handler the counts keep increasing, but to a certain extent. So apparently the gating takes "full effect" after a while (hundreds of counts at 4 MHz). It behaves much like analogic and not digital.  All the interest of the gating is to have a precise time gating of course. I use an interrupt but with he idea that the counter was stoppped precisely. If I rely only on interrupt it is likely that timing will be less accurate and even totally wrong in case the interrupt is delayed. 

I will try other ways. 

- Changing the prescaler / overflow values of timer 1 / maybe it does change the precision of the IRW firing ? 
- Using the repetition counter for a 16+8 bits counter to simplify and get rid of timer 2 / timer 3 chaining.

Otherwise I will try on a M0 board.

stevstrong, 2019-05-02:
I don't quite understand the scope of "gating". You just need to read two consecutive counter values in fixed time interval while the input signal is the clock, and take the difference of them.
If you use the systick IRQ each ms to read out the counter, you will have a difference up to 4000 at 4MHz.

mrguen, 2019-05-02:
Yes the problem is "fixed time interval" with a precision of 1 clock count if possible. IRQ's can be delayed. But good news. 

Today I tested with DMA request sent on Timer 1 update event and got exaclty the same result as when using the Enable MSM signal to gate timer 2 with timer 1. 

I thought the result was false because it reads 4000136 counts at 4 MHz. Since DMA gives exactly the same apparently gating based on Enable is ok (as explained RM008 p 340 MMS[2:0]: Master mode selection 001 Enable). 

The precision error must depend on something else, maybe the crystal...

For god's sake I will provide the DMA sketch here:  

// TimeBase 1 second generates DMA request to populate frequency from timers CNT
// Edge counter = TIM2 chained with TIM3 to form a 32 bits timer

// Finalement on obtient la même chose qu'avec le script basé sur l'esclavage ENABLE. !!
// Mais à la différence du script basé sur interruptions, on ne risque pas d'avoir un résultat faux à cause d'une IRQ en retard.

`#include "Streaming.h"
#include <libmaple/dma.h>

dma_tube_config dma_cfg;

unsigned int test = 33;
unsigned int LSB = 0;
unsigned int MSB = 0;
unsigned long lastCount = 0;
unsigned long count = 0;
unsigned long frequency = 0;

int ret = 1;

void dmaCompleteHandler() {

 }
 
void periodHandler() {

  delay(100);
  count = (MSB<<16) + LSB;

  if (count < lastCount) { // The 32 bits counter has overflowed
    lastCount -= 2^32;
  }
  
  frequency = count - lastCount;
  lastCount = count;
  
  Serial << "f: " << frequency  << endl; 
}

void DMA_Setup(void){
  
  dma_init(DMA1);
  
  // Transfer Timer 2 counts to LSB on Timer 1 Update event
  dma_cfg.tube_src = &(TIMER2->regs.gen->CNT); 
  dma_cfg.tube_src_size = DMA_SIZE_16BITS;
  dma_cfg.tube_dst = &LSB;
  dma_cfg.tube_dst_size = DMA_SIZE_16BITS;
  dma_cfg.tube_nr_xfers = 1;                              // Transfer one Word = 16 bits
  dma_cfg.tube_flags = DMA_CCR_CIRC;                      // Circular mode. 
  dma_cfg.tube_req_src = DMA_REQ_SRC_TIM1_UP;             // Request on Timer1 update event
  dma_cfg.target_data = 0;
  ret = dma_tube_cfg(DMA1, DMA_CH5, &dma_cfg);            // Channel 5 handles DMA_REQ_SRC_TIM1_UP

  // Transfer Timer 3 counts to MSB on Timer 3 Update event
  dma_cfg.tube_src = &(TIMER3->regs.gen->CNT); 
  dma_cfg.tube_src_size = DMA_SIZE_16BITS;
  dma_cfg.tube_dst = &MSB;
  dma_cfg.tube_dst_size = DMA_SIZE_16BITS;
  dma_cfg.tube_nr_xfers = 1;                            // Transfer one Word
  dma_cfg.tube_flags = DMA_CCR_CIRC;                    // Circular mode. 
  dma_cfg.tube_req_src = DMA_REQ_SRC_TIM3_TRIG;         // Request on Timer3 update event
  dma_cfg.target_data = 0;
  ret = dma_tube_cfg(DMA1, DMA_CH6, &dma_cfg);          // Channel 6 handles DMA_REQ_SRC_TIM3_TRIG

  dma_set_priority(DMA1, DMA_CH5, DMA_PRIORITY_VERY_HIGH);
  dma_set_priority(DMA1, DMA_CH6, DMA_PRIORITY_HIGH);
  dma_enable(DMA1, DMA_CH5); // Activates Timer 1 update DMA request
  dma_enable(DMA1, DMA_CH6); // Activates Timer 3 update DMA request

}

void TIMER_Setup(void)
{
    pinMode(PA0, INPUT);

    // TIME BASE *************************************************************************************
    timer_init(TIMER1);

    // See RM008 p. 399 "Using one timer to enable another timer"

    (TIMER1->regs).gen->CR2 = TIMER_CR2_MMS_COMPARE_OC1REF; // Master mode TRGO     
    (TIMER1->regs).gen->PSC = 7199;                         // Prescaler;
    (TIMER1->regs).gen->ARR = 10000;                        // Overflow 1 s

    (TIMER1->regs).gen->CNT = 8000;                         // No time wasted at start up

    timer_attach_interrupt(TIMER1, 0,  periodHandler);   

    timer_dma_enable_upd_req(TIMER1);

  
    // EDGE COUNTER LSB *********************************************************
    timer_init(TIMER2);
    (TIMER2->regs).gen->CR2 = TIMER_CR2_MMS_UPDATE ; // Master mode TRGO on overflow RM008 p 399    
    (TIMER2->regs).gen->SMCR = (TIMER_SMCR_ECE | TIMER_SMCR_MSM); // Clocked by extrenal signal mode 2 + MSM/overflow to chain counter 3

    // EDGE COUNTER MSB  *********************************************************
    timer_init(TIMER3);
    (TIMER3->regs).gen->SMCR = (TIMER_SMCR_SMS | TIMER_SMCR_TS_ITR1); //Triggered by ITR1 TRGO from counter 2 see Table 86 RM0008 and page 399 chaining two timers as prescaler

    timer_dma_enable_req(TIMER3,6);                         // Trigger DMA request enable

    // Start timers
    (TIMER2->regs).gen->CR1 = TIMER_CR1_CEN; // start edge count LSB
    (TIMER3->regs).gen->CR1 = TIMER_CR1_CEN; // start edge count MSB    
    (TIMER1->regs).gen->CR1 = TIMER_CR1_CEN; // start time base

}
void setup()
{
  DMA_Setup();
  TIMER_Setup();  
  Serial.begin(250000);
  while(!Serial); delay(10);
}

void loop ()
{

}`

stevstrong, 2019-09-15:
Any progress on this? Can be closed?

mrguen, 2019-09-15:
I could not get the precision I was looking for. The results depend on the method: interrupts, DMA or gating and there can be as high as 300 clocks overcount. So the precision is not what I was looking for. When using interrupts, clearly the handler is called "a while" after the interrupt is raised. But using DMA is not much better. When gating it is the same. The gating signal takes "a while". I have no idea if it is the Arduino implementation that generates so much parasite code or because of the delay that is exposed in the datasheet (RM008) p 335 and 336. Anyway I moved to STM32F404RC, since it has a 32 bits counter, it should be easier to get mor precision.

stevstrong, 2019-09-15:
I did not explore your code but are you sure that the measured signal is "constant"? I mean that the input signal can also have deviation from the ideal case, the number of pulses can maybe vary.
I lately experienced with chained timers to reach 32 bit precision and it worked very good. So this setup should not cause any headache.
I really don't understand how using the DMA should improve the results. Maybe it would be better to make a short drawing (even hand made) to explain your setup.
I also don't understand which form has the gating signal applied to timer 2. Does it have a 50% duty cycle? If yes, then the read out result should not be influenced at all by interrupt latency. A drawing would help here too.

mrguen, 2019-09-16:
I tried DMA so I was certain there was no delay due to retreiving low word and high word and even did not have to affect a variable since it was already in RAM. 
I am sorry at the moment I am busy but I will have to come back to this on the STM32F401RC. But not soon. If you have a working code with chained timers I would be interested. Thanks for the support.

rogerclarkmelbourne, 2020-02-06:
Closed because no current activity from participants

closed 2020-02-06 by rogerclarkmelbourne