rogerclarkmelbourne/Arduino_STM32 #631: Store millis() in a static variable making the sketch too big

xymopen, 2019-05-12:
[image]

Store `millis()` in a static variable throws 'Sketch too big;' error; store `millis()` in a global variable or a local auto variable is fine. This looks wired.

<details><summary>Build process dump</summary>
<p>

```text
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xymopen\AppData\Local\Arduino15\packages -hardware C:\Users\xymopen\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xymopen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xymopen\Documents\Arduino\libraries -fqbn=Arduino_STM32:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10809 -build-path C:\Users\xymopen\AppData\Local\Temp\arduino_build_780565 -warnings=all -build-cache C:\Users\xymopen\AppData\Local\Temp\arduino_cache_344575 -prefs=build.warn_data_percentage=75 -verbose C:\Users\xymopen\Documents\Arduino\sketch_may12a\sketch_may12a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xymopen\AppData\Local\Arduino15\packages -hardware C:\Users\xymopen\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xymopen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xymopen\Documents\Arduino\libraries -fqbn=Arduino_STM32:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10809 -build-path C:\Users\xymopen\AppData\Local\Temp\arduino_build_780565 -warnings=all -build-cache C:\Users\xymopen\AppData\Local\Temp\arduino_cache_344575 -prefs=build.warn_data_percentage=75 -verbose C:\Users\xymopen\Documents\Arduino\sketch_may12a\sketch_may12a.ino
Using board 'genericSTM32F103C' from platform in folder: C:\Users\xymopen\Documents\Arduino\hardware\Arduino_STM32\STM32F1
Using core 'maple' from platform in folder: C:\Users\xymopen\Documents\Arduino\hardware\Arduino_STM32\STM32F1
Detecting libraries used...
"C:\\Users\\xymopen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/include" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/stm32f1/include" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/stm32f1" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/usb_lib" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\cores\\maple" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\variants\\generic_stm32f103c" "C:\\Users\\xymopen\\AppData\\Local\\Temp\\arduino_build_780565\\sketch\\sketch_may12a.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\xymopen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/include" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/stm32f1/include" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/stm32f1" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/usb_lib" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\cores\\maple" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\variants\\generic_stm32f103c" "C:\\Users\\xymopen\\AppData\\Local\\Temp\\arduino_build_780565\\sketch\\sketch_may12a.ino.cpp" -o "C:\\Users\\xymopen\\AppData\\Local\\Temp\\arduino_build_780565\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\xymopen\\AppData\\Local\\Temp\\arduino_build_780565\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\xymopen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/include" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/stm32f1/include" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/stm32f1" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/usb_lib" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\cores\\maple" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\variants\\generic_stm32f103c" "C:\\Users\\xymopen\\AppData\\Local\\Temp\\arduino_build_780565\\sketch\\sketch_may12a.ino.cpp" -o "C:\\Users\\xymopen\\AppData\\Local\\Temp\\arduino_build_780565\\sketch\\sketch_may12a.ino.cpp.o"
C:\Users\xymopen\Documents\Arduino\sketch_may12a\sketch_may12a.ino: In function 'void setup()':

C:\Users\xymopen\Documents\Arduino\sketch_may12a\sketch_may12a.ino:2:18: warning: unused variable 'tick' [-Wunused-variable]

  static uint32_t tick = millis();

                  ^

Compiling libraries...
Compiling core...
"C:\\Users\\xymopen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-gcc" -c -g -x assembler-with-cpp -MMD -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/include" "-IC:\\Users\\xymopen\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/stm32f1/include" "-IC:\\Users\ [... cut ...] 2a.ino.elf"

Sketch uses 75168 bytes (114%) of program storage space. Maximum is 65536 bytes.
Global variables use 6336 bytes (30%) of dynamic memory, leaving 14144 bytes for local variables. Maximum is 20480 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Generic STM32F103C series.
```

</p>
</details>

I'm not sure whether it should be posted here or [stm32duino](http://www.stm32duino.com/). Feel free to move it.

rogerclarkmelbourne, 2019-05-12:
Duplicate issue

See the issue about Thread Safe Static’s

closed 2019-05-12 by rogerclarkmelbourne

xymopen, 2019-05-12:
Thanks for the renference.