rogerclarkmelbourne/Arduino_STM32 #633: Strange STM32F103C8T6 variant

Szekyy, 2019-05-13:
Dear Roger Clark,

It's  not an issue, just a question. I found a strange but very useful looks STM32F103 board on ebay. There is nothing on the internet about it.

[image]

... This is the cutie I'm talking about. It has so many different outputs. But I'm not sure if those can be used in the same time. I noticed that it (might) using the same pin for more than one output. Aaand that's all I found out. 

You can find those called:
STM32F103C8T6 "Core Board" or "Mini system board".

My question, can you tell me something about it?

Thank you in advance for your help, and sorry for my bad English.

ssieb, 2019-05-13:
That's a very interesting board layout.  All the port pins are grouped together but also broken out in combinations for different uses.  Some of the pins will be going to multiple places.
I found what appears to be a description of most of the pins and how they are linked:
https://www.airspringsoftware.com/image/s3-ap-southeast-1.amazonaws.com/a2.datacaciques.com/16/09/22/s14q19098bmp2dx3/8ffd3663e23f930f.jpg

Szekyy, 2019-05-13:
Wow, looks pretty good. Now I understand that. I found a picture like this before but couldn't read because of the low pixel count. Anyway, it's much clearer now. 

My main goal is to have a connection between my pc and this via WIFI and then transfer it to my drone with NRF24 :smile: 

Thank you for fast respond :upside_down_face:

ssieb, 2019-05-13:
After uploading it here, I see that something has been whited out in the top-left corner, but I've found the same image in multiple places.
Also here is some more info:
https://wiki.cuvoodoo.info/doku.php?do=export_xhtml&id=stm32f1xx
If you want wifi, you will need to add something else like one of the esp8266 boards that it has a connector for.

Szekyy, 2019-05-13:
Oooo, ure a god bro :heart_eyes:

zoomx, 2019-05-14:
I have this board. As @ssieb told, same pins are grouped.
On the other side there are pads to solder an microSD slot, a battery holder and some other chips.

There are some post about this board in the STM32duino forum.

Szekyy, 2019-06-07:
Hey. Back here! The Board just arrived. Everything works fine! But... Why do we need that Atmel704 on the bottom :open_mouth: ? Is it for memory control in case we use SDCard or FLASH memory? I have no idea :thinking:

Szekyy, 2019-06-07:
> Hey. Back here! The Board just arrived. Everything works fine! But... Why do we need that Atmel704 on the bottom  ? Is it for memory control in case we use SDCard or FLASH memory? I have no idea
@ssieb 
@zoomx 
any idea?

zoomx, 2019-06-07:
The "atmel 704" is an AT93C46 serial EEPROM. As I wrote before there are pads for soldering chips but they are not mandatory, only if you need them!

Szekyy, 2019-06-09:
[image]
So I started making a new layout drawing and thinking about to create a repository that contains information and examples for the board.

stevstrong, 2019-09-14:
I will close this because it seems to be solved.

closed 2019-09-14 by stevstrong