rogerclarkmelbourne/Arduino_STM32 #635: STM32F103 DMA

nikjayswal, 2019-05-17:
Hi i am not a professional Arduino programmer. I had searched on Google how to use DMA in stm32f103 and Arduino but doesn't find any examples also not finding any simple examples for stm32dunio site to learn how to use DMA for beginners. I want to use DMA for reading analogRead much fast as possible..... please provide some tutorial Examples of DMA how to use it.

ssieb, 2019-05-19:
DMA is not really a beginner topic, but here are some links to get you started.
libmaple DMA header file: http://docs.leaflabs.com/static.leaflabs.com/pub/leaflabs/maple-docs/latest/libmaple/api/dma.html
STM32 reference: https://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf
Section 13 for DMA
Section 11.8 for ADC DMA
Probably sections 14 and 15 for timer interaction with DMA.

You are unlikely to need DMA for ADC as there are other ways.  You could use a timer for triggering the conversion.  The library also has support for interrupts at the end of conversion, see https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/system/libmaple/include/libmaple/adc.h for details.

stevstrong, 2019-05-19:
Some examples:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/STM32ADC/examples/SingleChannelAtSampleRateCircularBuffer/SingleChannelAtSampleRateCircularBuffer.ino

ssieb, 2019-05-19:
Very nice!  I didn't notice in the Arduino IDE that the examples list scrolled far past the end of what showed.  That's a perfect example.

nikjayswal, 2019-06-13:
Does this example work in void loop " https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/STM32ADC/examples/SingleChannelAtSampleRateCircularBuffer/SingleChannelAtSampleRateCircularBuffer.ino "

rsonnicksen, 2019-08-19:
I would like to use DMA to transfer serial data from a GPS receiver to memory so I can parse after all the data is received.  So, I need help setting up DMA, and I also would like to trigger my parsing routine when the LineIdle interrupt for the USART fires indicating the GPS receiver is done sending "this" burst of data.  I'm definitely a newbie to C and Arduino compiler so very confused if, where and how the USART interrupts are handled for the STM32F102 Blue Pill board.  I've found __irq_usart2 routine which I believe is interrupt routine for USART2 interrupt, and this routine calls usart_irq which seems to read bytes from the usart and push them into a ring buffer (which is itself confusing as it looks like an array, but isn't delared as an array.).  There are so many pieces of this code which are beyond my level so I'm really needing some guidance to know how/what to do to read a usart using DMA.  FYI, I am trying to improve on the code from a fellow named Joop Brokking out of the Netherlands who has migrated an Arduino Uno based flight controller (quad copter) to the STM32 platform.  See http://www.brokking.net/.  I appreciate any constructive or guiding comments.  Thanks in advance.  Randy

stevstrong, 2019-08-22:
@rsonnicksen ,
it is an overkill to use DMA for receiving serial data over UART.
DMA is useful for collecting data when the time is critical. But for UART it is not the case, the current implementation of the receiving part is working very well and is very robust. Do you have a problem with that? If yes, then open a new issue.
You could eventually change the actual ISR for UART to track the number and/or value of the incoming bytes.
You can eventually write your own UART ISR and replace the default one with yours.

rsonnicksen, 2019-08-22:
**@stevstrong**,
Thanks for your response, but I didn't find it helpful.  The information I'm looking for is:
Where is the UART ISR for the STM32F103c board (blue pill)? (What file/folder).
How do I enable DMA for the USART?
How do I enable the LineIdle interrupt, and disable the RXNE interrupt? (without editing the MapleLib).
How do I set the pointer to "my" USART irq_hander?
Thx - Anyone?

stevstrong, 2019-08-22:
If you are newby to C and Arduino compiler and you are anyway confused with a simple thing like UART, I strongly suggest **not** to get involved with DMA, which is a complicated beast.
Just check in your main loop for the number of received bytes with
`int rcvd = Serial2.available(); // gives the number of bytes available in receive buffer`
Then you can parse the received data.
I still don't understand why you want to use DMA. The standard Arduino serial API functions should be enough to detect reception of "this" burst of data

nikjayswal, 2019-08-24:
There is no option showing on DMA examples to enable and disable DMA transfer. How to enable and disable DMA transfer ?

nikjayswal, 2019-08-25:
On example " myADC.startConversion();  " to start DMA but when i edit to " myADC.stopConversion();  " it gives me error. How to disable DMA transfer ?

stevstrong, 2019-08-25:
There is no such a function to stop DMA, but you are welcome to implement it and send a PR.

JiriBilek, 2019-08-25:
@nikjayswal : 
1. I am convinced you don't need to call `myADC.startConversion();` in SingleChannelAtSampleRateCircularBuffer example;
2. All you need to do to stop the ADC is to take out the trigger clock from the ADC module, i.e. `myADC.setTrigger(ADC_EXT_EV_SWSTART);`
3. My advice: you should really read the STMF103 Reference manual carefully. You should first know how the ADC and DMA work before using them. Otherwise, I am afraid, you are wasting your time ;-)

closed 2019-08-25 by nikjayswal