rogerclarkmelbourne/Arduino_STM32 #636: 32-bit linux dfu-util fails because it is 64-bit

manitou48, 2019-05-25:
zip file containing the STM32 files has a 64-bit executable in linux/dfu-util/dfu-util so dfu-util fails on a 32-bit linux system

 **sketchbook/hardware/Arduino_STM32-master/tools/linux/dfu-util/dfu-util: cannot execute binary file: Exec format error**

ag88, 2019-06-19:
i'd like to suggest you could build this from source
http://dfu-util.sourceforge.net/
i've been thinking about committing a PR for this but i found that it has quite a few shared library dependencies. i did a built from a 64 bit environment with 32 bits libs, and my libs may be different from that used on an actual 32 bits distribution and it may not be compatible with that kernel.

stevstrong, 2019-09-15:
I have never worked with Linux.
Who can help here?

ag-123, 2019-09-16:
hi steve, my opinion is that it would be better for the users of 32 bit linux to built the dfu-util binary themselves.

nevertheless i've built a 32bit version and attached here.
[dfu-util.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/3614709/dfu-util.zip)

there is a 00readme.txt file in there place it in the same folder with linux's 32 bit dfu-util.

mrstanlez, 2020-04-20:
hi, I want to ask you for any info about linux 64bit, arduino ide and maple mini uploading sketch direct through USB. In win10 it works perfect.
Any sugestion ?
Thank you

ag88, 2020-04-21:
it should work in linux, the binary that is shipped in the core is 64bit. 32 bits are less common these days but would apply only if you run an os that boot in 32 bit mode.