rogerclarkmelbourne/Arduino_STM32 #637: I2c_scanner_wire cannot loop more than once without a I2C device

1031063068, 2019-06-02:
A phenomenon where the Stm32duino Wire sample program I2c_scanner_wire cannot loop more than once without a I2C device, and the board is BluePill

enesbcs, 2019-06-11:
Much deeper. When i try i2c scan without connected I2C devices OR I2C devices without proper pullup resistors, sketch is not responding anymore until reset.

rogerclarkmelbourne, 2019-06-12:
This issue has been raised before and may be a limitation of the STM32's hardware I2C

Try using the official core, it may work better since it uses the official HAL to interface with the hardware, and may have better error handling.

enesbcs, 2019-06-12:
Thank you for your reply, but unfortunately the stm32f103ret6 with 512k flash is not supported at the official core as i see, so it is not an option to me.

rogerclarkmelbourne, 2019-06-12:
it would be easy to make a new variant of the official core to use that chip, especially if you don't need to use the DAC and a few other things which it has and the F103C8 doesn't

Post an issue on the official core and I'm sure Frederic and the guys can point you in the correct direction

enesbcs, 2019-10-02:
> it would be easy to make a new variant of the official core to use that chip, especially if you don't need to use the DAC and a few other things which it has and the F103C8 doesn't

The stm32f103ret6 is now supported by the official core, i can use it for a simple Blink example...  and my sketch compiles with it ( https://github.com/enesbcs/ArduinoEasy ) , but it never starts up, and prints nothing to serial to diagnose it. (and somehow also kills the onboard bootloader) 
With your core, my sketch works OK, so i think i can live with this i2c problem. :)

stevstrong, 2019-10-02:
You may try #679 to see if it helps.

enesbcs, 2019-10-03:
Thanks i will try it!

TheKikGen, 2019-10-04:
I use it with success currently.  It is far better than  the current one, notably on the bus state management.  Another point is that is not easy to migrate to the "official" core currently...

enesbcs, 2019-10-05:
I can confirm that this fix eliminates the I2C scanning problem. Thanx!

fpistm, 2019-10-05:
@enesbcs 

> The stm32f103ret6 is now supported by the official core, i can use it for a simple Blink example... and my sketch compiles with it ( https://github.com/enesbcs/ArduinoEasy ) , but it never starts [...]

I will be glad to help to understand why.
I already have an idea but could yo provide the sketch which never start 😉

enesbcs, 2019-10-05:
> I will be glad to help to understand why.
> I already have an idea but could yo provide the sketch which never start wink

The sketch is available on the github, here: 
https://github.com/enesbcs/ArduinoEasy

There are a lot of possibility, first i thought that the HAL_FLASHEx_Erase (src/Misc.ino) somehow erases the bootloader, but i've never seen any serial.println() output on the serial console, after uploading the sketch. Honestly this is not my sketch, i've just ported the original to STM32F103. And it works fine with Roger's core.

stevstrong, 2019-10-26:
I will close this because the original issue has been solved.
For the issue related to the STM official core should be posted on that repository.

closed 2019-10-26 by stevstrong