rogerclarkmelbourne/Arduino_STM32 #638: Strange issue while using breadboard and bluepill (IIC)

effeci, 2019-06-10:
Hello everyone,
I'm using this same framework and usb bootloader.
The issue is that i'm trying to execute an IIC scan . If the bluepill linked to the display with jumpers everything is ok while if i connect it on breadboard it just hangs. this looks to me really strange. Anyone has any clue?
Thanks

GitMoDu, 2019-06-17:
Possible duplicate #637

stevstrong, 2019-10-26:
The I2C library has been recently updated, so please check again if the issue still persists.

TheKikGen, 2019-10-26:
Check your pullup resistors, and your connections.  Use 2x4,7K for clock and data.
Then use sample provided with the library to test your proto.
The I2C (Wire_slave) lib works perfectly with Bluepill.

david314, 2019-11-30:
I am having the same problem. The Wire.h library produces a strange glitch on endTransmission on my Arduino Maple. Instead of generating a stop on the bus, an additional shortened 10th clock pulse is generated and after that the device stops working. If the clock frequency is reduced below ~60kHz the behavior changes. Very strange.
[image]

dewhisna, 2019-11-30:
First off, what "device" are you trying to communicate with?  And what does your code look like?  If I'm reading these bits correctly, you are trying to do a write operation of zero bytes to a device at address 0x01, which is an invalid I2C address and is used as a placeholder in the I2C code until a valid target address is specified.  You mention calling `endTransmission`, but did you ever call `beginTransmission` and give it a target address??  What does your code look like??

The extra bit cycle at the end appears to only be part of the ACK/NACK portion and is a NACK because your device (whatever it is) doesn't acknowledge the transmission -- which it probably shouldn't as I doubt its address is 0x01, since that's a reserved I2C address in the standard (https://www.i2c-bus.org/addressing/).

You also say:
> If the clock frequency is reduced below ~60kHz the behavior changes.

But you don't say how or what it has to do with anything.

As Johnny Five would say: "Need more input".

david314, 2019-11-30:
I have different setups, e.g. one with a BMp280 at 0x76. SlaveWire and SoftWire have no issues finding the devices connected, but the i2c_scanner_wire.ino script shows the problem with the Wire.h library. Don't ask me why the script starts at 0x01, but in any case at frequency close to 60kHz it may run for a few cycles before showing the erroneous behavior. In any case, the tenth clock cycle is out of spec and makes no sense.

`// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//

#include <Wire.h>

//use IIC2
//TwoWire WIRE2 (2,I2C_FAST_MODE);
//#define Wire WIRE2

void setup() {

  Serial.begin(115200);
  Wire.begin();
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");

  delay(5000);           // wait 5 seconds for next scan
}
`

dewhisna, 2019-12-01:
@david314  I have something for you to test.  I think the issue may be with the error handler interrupt for the I2C code.  Comparing with the STM32Cube code, it looks like the master should command a stop if it receives a `SR1_AF` (acknowledge fail) error.

I've created a Git bundle for you to test.  I had to zip it to get GitHub to attach it here, but if you unzip the attached fix_i2cmaster_StopOnNACK.zip and place the `.bundle` file one level above your repo clone, and then in your Git bash command prompt from inside the `Arduino_STM32` folder of your repo clone, first make sure you have checked out the `master` branch from this repo and that it's current.  Then you should be able to run:

```
git pull --ff-only ../fix_i2cmaster_StopOnNACK.bundle fix_i2cmaster_StopOnNACK:master
```

This will fast-forward merge my test commit onto the master branch of your local checkout.  Run `git log` after you do that to make sure you see the new commit at the top of the list and/or run `git status` and it should show you are one commit ahead of the upstream master.

Then rebuild your code and see if that fixes your glitch.  If it does, I'll create a pull-request to get it fixed here in this repo.

[fix_i2cmaster_StopOnNACK.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/3907857/fix_i2cmaster_StopOnNACK.zip)

stevstrong, 2019-12-01:
Insert 
`i2c_stop_condition(dev);`
before this line
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/system/libmaple/include/libmaple/i2c.h#L282

david314, 2019-12-01:
I am travelling the next week, but I will try it asap. Are your comments the same fix?

stevstrong, 2019-12-01:
I am extending what @dewhisna explained, in case of error the line should be stopped with a stop condition before disabling and enabling again the interface.

dewhisna, 2019-12-01:
@stevstrong  No, you don't want to always do a stop condition there.  This code also has to work for the slave as well as the master.  The master has to control the stop condition.  You'll want to use the commit in the bundle in the zip I posted above where it only sends the stop in the I2C error interrupt handler for the explicit condition of Master-Only receiving a SR1_AF (acknowledge failed) condition.

If the bundle is too confusing for people, I can push the branch to my fork.  That may make it easier for people to merge and try who aren't familiar with bundles.  It was late last night (after midnight) and for some reason the bundle seemed easier at the time...  I'll just go ahead and push my branch to my fork.  As I do think we'll need to merge it.  I'll even open the pull-request -- just maybe not merge it until @david314 gets a chance to try it, since he has a scope setup for it.

david314, 2019-12-01:
Ok, I will use your fork and post my results. Unfortunately, I won't be back at my desk before 10th of December to look at the waves again. Anyway, I can check if the scripts runs on my Maple or not.

david314, 2019-12-02:
I have updated the four lines 785..789 in i2c.c and can confirm that the script works now with my setup for frequencies up to 1MHz.

dewhisna, 2019-12-02:
@david314  Great!  After digging through ST's code for I2C, I figured it would solve it.  So @stevstrong you can probably go ahead and merge #722 now.

PR #722 (2019-12-03): I2C Master should send a STOP when receiving NACK error (SR1_AF)
This is the PR that fixes #638 and is equivalent to the bundle I posted there at [fix_i2cmaster_StopOnNACK.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/3907857/fix_i2cmaster_StopOnNACK.zip)

The NACK should only be sent by the master and the only one missing was the error case in the this interrupt handler for the AF (acknowledge failed) case.  The other stop conditions are handled by the main interrupt routine.

Adding a stop to all paths would be wrong because it can actually cause a restart condition with th [... cut]
closed 2019-12-03 by stevstrong