rogerclarkmelbourne/Arduino_STM32 #639: usb problem on custom board

sensestage, 2019-06-12:
With the forum not active right now, I'm not sure where to turn for support. (I could register to the forum, but not start a new topic).

I have made a definition for a custom board, by making a variant of the closest board (Discovery407VG) and then adjusting the clocks.

I'm trying to get USB working - this might be a hardware problem, but I want to be sure that I have the firmware set up right.

In `stm32f4xx_hal_conf.h` I've uncommented:
```#define HAL_PCD_MODULE_ENABLED```

And I have uploaded a minimal example (see below), using the stlink upload method.
I've set:
- c runtime lib (newlib nano (default))
- usb speed (low/full speed)
- usb support (cdc (generic serial supersede usart))
- usart support (enabled generic serial)

example code:
```
#define mySerialPort SerialUSB

void setup() 
{
  mySerialPort.begin(115200);
  mySerialPort.println("Serial Port Example");
  mySerialPort.println("just counting up");
}

int Counter=0;

void loop() 
{
  mySerialPort.println(Counter); 
  Counter++;
  delay(1000);
}
```

With `dmesg` I get the following error:
```
[122844.939027] usb 1-1.2:  new full-speed USB device number 77 using ehci-pci
[122845.355030] usb 1-1.2: device not accepting address 77, error -32
[122845.439050] usb 1-1.2: new full-speed USB device number 78 using ehci-pci
[122845.855017] usb 1-1.2: device not accepting address 78, error -32
[122845.855411] usb 1-1-port2: unable to enumerate USB device
```
Indicating that something goes wrong in enumerating the device.

Again, this may be a hardware issue (I've used a design similar to the [OWL board](https://hoxtonowl.com/mediawiki/index.php/Schematics) for the USB connection), but I need to know if I set the firmware options right.

fpistm, 2019-06-12:
@sensestage 
It seems you submit an issue on the wrong place.
This should be here:
https://github.com/stm32duino/Arduino_Core_STM32
Anyway, to answer your questions.
To enable USB support, you only have to select _usb support (cdc (generic serial supersede usart))_
this enable `HAL_PCD_MODULE_ENABLED`. So you don't need to uncomment it in the `stm32f4xx_hal_conf.h`.
If you clock properly at 48 MHz the USB then it should work.

Finally, the forum is operational since STM announcement: See http://stm32duino.com/viewtopic.php?f=16&t=4671&p=54422 
Thanks again @rogerclarkmelbourne ;)

sensestage, 2019-06-24:
thanks, I'll switch to the forum. I was confused as I could not create a new topic in the appropiate category, but after an intro-post in the new users subforum I can.

closed 2019-06-24 by sensestage