rogerclarkmelbourne/Arduino_STM32 #642: establishContact

mrguen, 2019-06-17:
Hello,

While using the sketch example "EEPROM_example.ino" I could not see the HELP_MSG printed on Serial in the setup() function.

I looked at the  sketch named "SerialCallResponse.ino" : there is a function that is called in setup and that allows to have a functionnal Serial.print afterwards (in setup()). 

void establishContact() {
    while (Serial.available() <= 0) {
        Serial.write('A');   // send a capital A
        delay(300);
    }
}

So, I would like to know if there is a bug or what? Is there another way to corretly initialize the Serial Monitor?

GitMoDu, 2019-06-17:
Since BluePill has USB emulated, you should wait for first serial connection before starting any other serial stuff. Tipically, this is done like:

> setup() {
> Serial.begin(SERIAL_BAUD_RATE);
> 	while (!Serial)
> 		;
> }

fpistm, 2019-06-17:
FYI, documented here:
https://www.arduino.cc/reference/en/language/functions/communication/serial/ifserial/

mrguen, 2019-06-17:
Thanks for the info. I tried to replace the establishContact();  with while (!Serial) ; but this is not working...

stevstrong, 2019-06-17:
`while (!Serial) ; delay(10);`
should work.
The reset always causes the USB serial port to be re-enumerated by the host (PC), this is normal.
The serial monitor should be closed and opened after each reset.

mrguen, 2019-06-17:
Thanks,

while (!Serial) ; delay(10); yes this is working

The board canno't be re-enumerated if ever the reset happens when the Serial monitor is opened. So apparently it needs to be closed before resetting the board and re-opened.

joej970, 2019-08-22:
Hi!
`Serial.begin(115200);
  while (!Serial);
  delay(10);`
does not work for me. Though `Serial` is true, because I do get past while statement and then into loop().
I am using **stm32duino 7kb bootloader**, but under upload methods I select Mapple bootloader 2.0 at PORT **5** but I think that these two are the same bootloaders. 
Anyway, If I flash 20kb generic bootloader and upload scketch through upload method> STM32CubeProg Serial at PORT **3**, I can read Serial Monitor (using USB to Serial adaptor at PORT **3**)

Is Serial.begin() correct call? Should it maybe be Serial1, or SerialUSB or SERIAL_PORT_MONITOR.begin();?

mrguen, 2019-08-22:
@joe970 I Think there is a confusion between STM32duino and Arduino_core_STM32
Personnaly when using STM32duino core I have these download options :  
* STM32duino bootloader
* Serial
* STLINK
* BMP 
* JLINK
* HID bootloader 2

The STM32CubeProg option is with Arduino_core_STM32.

joej970, 2019-08-25:
@mrguen You are right. It works now. Thank you.

stevstrong, 2019-09-14:
I will close this as it seem to be solved.

closed 2019-09-14 by stevstrong