rogerclarkmelbourne/Arduino_STM32 #643: Big big number Union not working on Maple Mini

mrguen, 2019-06-19:
Hello,

I was trying to use a Union to store 32 bits data in EEPROM and could not make it (the UNION) work even though it is published on 
http://stm32duino.com/viewtopic.php?f=28&t=3321&p=42256&hilit=UNION&sid=a7c9c5040b303108d1cad2487794a79d#p42256

The board is a Baite Maple Mini clone with the bootloader 2.0

What I see is that it does work for small numbers but not for large numbers. Here is the text sketch and output

*****************************************************************
void setup() {
  
Serial.begin(57600);
while (!Serial);
delay(10);
  
new_name.big_number = 123.32;

Serial.println(new_name.big_number , HEX);
Serial.println(new_name.low_order_byte , HEX);
Serial.println(new_name.high_order_byte , HEX);

new_name.big_number = 150000000.0;

Serial.println(new_name.big_number , HEX);
Serial.println(new_name.low_order_byte , HEX);
Serial.println(new_name.high_order_byte , HEX);

}
****************************************************

Output:

123.3199999999999931
AE14
E147
150000000.0000000000000000
0
0

ssieb, 2019-06-20:
A double is 64 bits, the example you point to is using float which is 32 bits.

mrguen, 2019-06-21:
Ok, Is there a reference of datatypes on STM32duino / STM103F ?

stevstrong, 2019-06-21:
For example: http://www.keil.com/support/man/docs/armcc/armcc_chr1359125009502.htm

ssieb, 2019-06-21:
It's not really specific to this device.  Those are standard types.  There does tend to be some variance between platforms in the defined int size, but floats and doubles are always the same.
If you really need to use a double, you will also need to have 4 uint16 variables to match it.

stevstrong, 2019-06-21:
It is specific.
And even more specific would be this: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0491c/Babfcgfc.html
Just try it out.

rogerclarkmelbourne, 2019-06-21:
There are no standard types.

The underlying implementation is processor dependant unless you specifically use uint32_t etc

And even if you specify uint32_t etc the byte order will vary from processor to processor

mrguen, 2019-06-21:
Thanks Roger, I searched on the RM008 by keyword "data type" and found nothing... how do you determine the correct type for a given MCU?

mrguen, 2019-06-21:
Meanwhile I used this instead of a UNION

void shortsToDouble( unsigned short myShorts[2], double *myDouble) {
  *myDouble = myShorts[0]<<16 | myShorts[1];  //Big-endian
}

void doubleToShorts( double * myDouble, unsigned short myShorts[2] ) {  
  // Using a long because shifting operations are not allowed on double
  long dummyLong = * myDouble;
  myShorts[0] = (unsigned short) (dummyLong  >> 16) ;  
  myShorts[1] = (unsigned short) (dummyLong & 0xFFFF);
}

ssieb, 2019-06-21:
@stevstrong, sorry I wasn't referring to your comment and my comment was very badly worded.  I was just saying that float and double are almost always standard types independent of the architecture.

mrguen, 2019-06-21:
Ok, finally I used this to store unsigned shorts in flash

void shortsToFloat( unsigned short myShorts[2], float *myFloat) {
  *myFloat = myShorts[0]<<16 | myShorts[1];  //Big-endian
}

void floatToShorts( float * myFloat, unsigned short myShorts[2] ) {  
  // Using a long because shifting operations are not allowed on double
  long dummyUnsignedInt = * myFloat;
  myShorts[0] = (unsigned short) (dummyUnsignedInt  >> 16) ;  
  myShorts[1] = (unsigned short) (dummyUnsignedInt & 0xFFFF);
}

mrguen, 2019-06-21:
Ok, finally I used this to store float "converted" to unsigned shorts in flash

void shortsToFloat( unsigned short myShorts[2], float *myFloat) {
  *myFloat = myShorts[0]<<16 | myShorts[1];  //Big-endian
}

void floatToShorts( float * myFloat, unsigned short myShorts[2] ) {  
  // Using unsigned int because shifting operations are not allowed on float
  unsigned int dummyUnsignedInt = * myFloat;
  myShorts[0] = (unsigned short) (dummyUnsignedInt  >> 16) ;  
  myShorts[1] = (unsigned short) (dummyUnsignedInt & 0xFFFF);
}

rogerclarkmelbourne, 2019-06-21:
The EEPROM library is very basic

You could modify it to allow reading and writing of an arbitrary length buffer.

I'm working on a rather complex embedded open source radio project in my spare time, and I've written all the non volatile storage stuff from scratch

I'm using a struct to store all the settings and I write it to EEPROM by passing a pointer to the struct memory and the length of the struct

But also wrote all the underlying EEPROM libraries as well, from scratch, and they just take a pointer to a buffer to either write to EEPROM or read from EEPROM into the buffer pointer

https://github.com/rogerclarkmelbourne/GD-77/blob/master/GD-77_new_firmware/firmware/source/functions/fw_settings.c

mrguen, 2019-06-21:
Ok, thanks it's a better solution. I am using a class object. Maybe I could serialize its properties it in the same way without passing by a struct?

rogerclarkmelbourne, 2019-06-21:
A class generally contains methods as well as properties, so I'd be cautious of using a class instead of a struct

You'd need to check the actual size e.g. sizeof( your_class)  to see if it contains anything other than the properties.

Effectively all you do is cast your data, whatever it is, to a buffer of  uint8_t *  , i.e bytes

You can even do this with your float of whatever it was you were trying to serialise

However, I can't remember if the underlying EEPROM library supports  byte buffer read and write.

But it would not be difficult to change the library code to do this.

Serialising like this is safe as long as you always read and write to the data with the same device, but does not work, for example if you had both a X86 CPU and a STM32F103 writing and reading to the same EEPROM, since their underlying representation of the higher level types e.g. float would be different.

mrguen, 2019-06-21:
@rogerclarkmelbourne  
Inspired by your example I wrote this code. It is strange because it works for most of the data but not for ".fr" that is a float 

struct clockParameters
{
  byte id;        
  boolean ac;    
  float fr;      
  char un[1];     
  bool in;        
  unsigned int st;        
  byte dr;         
  byte ph;         
} ;

clockParameters clockParam = {0,0,0,"",0,0,0,0};

void eepromWrite(unsigned short address, unsigned short *data, unsigned short size){
// Size is in bytes, variables are stored in 2 bytes

  // Compléter
  unsigned short status;

  for (int i = 0; i < size / 2; i++) {
    status = EEPROM.write(address+i, *(data + i * 2));  
    if (status > 0) {
      Serial.print("Flash write error on data number ");
      Serial.print(i);
      Serial.print(": ");      
      Serial.println(status, HEX);
    }
  }
}

void eepromRead(unsigned short address, unsigned short *data, unsigned short size){
  unsigned short status;
  for (int i = 0; i < size / 2; i++) {
    status = EEPROM.read(address+i, data + i * 2);  
    if (status > 0) {
      Serial.print("Flash read error on data number ");
      Serial.print(i);
      Serial.print(": ");      
      Serial.println(status, HEX);
    }
  }
}

void saveOneClock(Clock &oneClock, byte slot) {
  unsigned short addressWrite = flashClockBaseAddress + slot * sizeof(clockParameters) ;
  unsigned short status;

  clockParam.id = oneClock.id;
  clockParam.ac = oneClock.ac;
  clockParam.fr = oneClock.fr;
  memcpy( clockParam.un, oneClock.un,1);
  clockParam.in = oneClock.in;
  clockParam.st = oneClock.st;
  clockParam.dr = oneClock.dr;
  clockParam.ph = oneClock.ph;

  eepromWrite(addressWrite, (unsigned short*)&clockParam, sizeof(clockParameters));  
}

void loadOneClock(Clock &oneClock, byte slot) {
  unsigned short addressRead = flashClockBaseAddress + slot * sizeof(clockParameters) ;
  
  eepromRead(addressRead, (unsigned short*)&clockParam, sizeof(clockParameters));  

  oneClock.id = clockParam.id;
  oneClock.ac = clockParam.ac;
  oneClock.fr = clockParam.fr;
  memcpy( oneClock.un, clockParam.un,1);
  oneClock.in = clockParam.in;
  oneClock.st = clockParam.st;
  oneClock.dr = clockParam.dr;
  oneClock.ph = clockParam.ph;

}

stevstrong, 2019-09-14:
Is this issue still to be solved?

mrguen, 2019-09-15:
Yes

stevstrong, 2019-09-15:
Solved.

closed 2019-09-15 by stevstrong