rogerclarkmelbourne/Arduino_STM32 #645: Spi to 8 bit parellel interface

nikjayswal, 2019-07-03:
How to drive 8 bit parellel interface RGB DISPLAY using SPI. ?
[image]

stevstrong, 2019-07-03:
You cannot drive it with SPI. Instead, use 8 bit of GPIOs.

Please use the main forum for further requests/issues: www.stm32duino.com.

nikjayswal, 2019-07-08:
Can't register to stm32dunio it giving error of questions

Question:
On an Arduino Uno, what pin is the LED connected to (include the D in front of the number):

Answer : D13 

But showing wrong answer

nikjayswal, 2019-07-08:
I had bought 74HC164 SPI to Parellel interface but i don't know which pins should i connect them 

Does this work to drive 8 bit parellel interface TFT using SPI ?

stevstrong, 2019-07-08:
Arduino Uno digital pins have only numbers, no letters.

No need for external components, there are ready 8 bit parallel libraries to drive the TFT display.

nikjayswal, 2019-07-08:
8 bit parellel takes too much pins

closed 2019-07-09 by nikjayswal

ssieb, 2019-07-09:
The SPI to Parallel chip will work if you only want to do writes, but you will also need at least one other line to the LCD board to tell it when the data is ready.