rogerclarkmelbourne/Arduino_STM32 #646: Adafruit_GFX_AS.cpp : error: 'textsize' was not declared in this scope

Rimel300, 2019-07-06:
I am new in stm32duino. Please advice me . How to solve this problem?

Arduino: 1.8.9 (Windows 7), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STLink, 72Mhz (Normal), Smallest (default)"

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp: In member function 'int16_t Adafruit_GFX_AS::drawUnicode(uint16_t, int16_t, int16_t, int16_t)':

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp:142:9: error: 'textsize' was not declared in this scope

     if (textsize == 1) drawFastHLine(x, pY, width+gap, textbgcolor);

         ^

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp:149:11: error: 'textsize' was not declared in this scope

       if (textsize==1){

           ^

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp:173:7: error: 'textsize' was not declared in this scope

   pY+=textsize;

       ^

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp:175:20: error: 'textsize' was not declared in this scope

 return (width+gap)*textsize;        // x +

                    ^

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp: In member function 'int16_t Adafruit_GFX_AS::drawCentreString(char*, int16_t, int16_t, int16_t)':

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp:249:15: error: 'textsize' was not declared in this scope

     len = len*textsize;

               ^

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp: In member function 'int16_t Adafruit_GFX_AS::drawRightString(char*, int16_t, int16_t, int16_t)':

C:\Users\Dev\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.6.29\libraries\Adafruit_GFX_AS\Adafruit_GFX_AS.cpp:300:15: error: 'textsize' was not declared in this scope

     len = len*textsize;

               ^

exit status 1
Error compiling for board Generic STM32F103C series.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ssieb, 2019-07-06:
According to the readme, you shouldn't be using this library anyway.
The problem here is that there was a recent change to the parent Adafruit_GFX library that split the textsize variable into x and y variants.  This library will need to be updated to account for that.

Rimel300, 2019-07-06:
Thanks , For your advice, But I am not clear this.

Rimel300, 2019-07-06:
I want to run **STM32-o-scope** or example 
[image]
 But It gives error to compile, 
error: 'textsize' was not declared in this scope

stevstrong, 2019-07-06:
Please check this post: https://www.stm32duino.com/viewtopic.php?f=14&t=4733

Most important, take into account the suggestion of David Prentice here: https://www.stm32duino.com/viewtopic.php?f=14&t=4733&start=10#p55062

Rimel300, 2019-07-06:
Thanks (stevstrong), 
I don't find any solution . Can you ?
mention the solution ?

Rimel300, 2019-07-06:
> According to the readme, you shouldn't be using this library anyway.
> The problem here is that there was a recent change to the parent Adafruit_GFX library that split the **### textsize variable into x and y** variants. This library will need to be updated to account for that.

Have I need to change in code or library ? if so, where ?

sorry for my bad english!!!!

stevstrong, 2019-07-06:
You have to change any occurrence of "Adafruit_GFX_AS" to "Adafruit_GFX" in ILI9341 lib files and in your sketch.

Rimel300, 2019-07-07:
> 

**I have to change some lib..
That is here  ........\/
                           \/''''''''''''''''''''''''''''''''**
\/''''''''''''''''''''''''''''''''''
//#include "Adafruit_ILI9341_STM.h"
//#include "Adafruit_GFX_AS.h"
\/''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
#include "Adafruit_ILI9341.h"           //     <<<<<<<<<<<<<<
#include "Adafruit_GFX.h"            //<<<<<<<<<<<<<
// Be sure to use the latest version of the SPI libraries see stm32duino.com - http://stm32duino.com/viewtopic.php?f=13&t=127
#include <SPI.h>
...
...
...
 **Also change Some pins Like..............**.
\/
\/

// Additional  display specific signals (i.e. non SPI) for STM32F103C8T6 (Wire colour)
#define TFT_DC      PB0      //   (Green) 
#define TFT_CS      PB1      //   (Orange) 
#define TFT_RST     PA2      //   (Yellow)
\/\

// Create the lcd object.........................................\/
//Adafruit_ILI9341_STM TFT = Adafruit_ILI9341_STM(TFT_CS, TFT_DC, TFT_RST); // Using hardware SPI
**I use...............................................\/''''''''''''''''''''''''''''''**
Adafruit_ILI9341 TFT = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST); // Using hardware SPI

// LED - blinks on trigger events - leave this undefined if your board has no controllable LED
// define as PC13 on the "Red/Blue Pill" boards and PD2 on the "Yellow Pill R"
#define BOARD_LED PC13 //PB0
\/
\/

#define TFT_LED        PC13     // Backlight 
#define TEST_WAVE_PIN       PB6     //PB1 PWM 500 Hz 

ohohohohooooohhhhhhhhhhhh
:( {}:( {}:( {}:( {}:( {}:( {}:( {}:( {}:( {}:( {}:( {}:( {}:( {}:( {}
My stm32 o scope work  !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
;-);-);-);-);-);-);-);-);-);-);-);-);-);-);-);-);-);-);-);-);-)
But My touch screen don't work. May be I have to

Rimel300, 2019-07-07:
Change  Calib????????  ...............................\/......................................
.........................................................................\/
 // #define NVRam register names for the touch calibration values.
#define  TOUCH_CALIB_X 0  //  ''''''  ..........\/''''''''''''''''''''''
#define  TOUCH_CALIB_Y 1//??????????????????
#define  TOUCH_CALIB_Z 2//???????????????????
**// I had change (Time1.h) in lib folder!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!1
// Time library - https://github.com/PaulStoffregen/Time
#include "Time.h" //If you have issues with the default Time library change the name of this library to Time1 for example.
#define TZ    "UTC+1"

stevstrong, 2019-09-15:
Is this issue still actual or it can be closed?

stevstrong, 2019-12-27:
I will close this because of lack of activity form the OP and is not related to the core functionality.

closed 2019-12-27 by stevstrong

bhall66, 2020-07-22:
stevestrong is correct about the ILI9341 library references to Adafruit_GFX_AS:  changing all refs in the ILI9341 library from Adafruit_GFX_AS to Adafruit_GFX solves the textsize compile problem.   

What is the reason that the latest version of the ILI9341 library still depends on Adafruit_GFX_AS?

jtannenba, 2020-08-10:
It fixed textsize in this example, but caused other errors. I, also am trying to compile STM32-O-Scope and these errors are still here

stevstrong, 2020-08-11:
Please post the errors as text here.

jtannenba, 2020-08-11:
Sorry, once I found that the older version of the GFX library fixed my issue, I dropped it.   Thank you