rogerclarkmelbourne/Arduino_STM32 #647: STM32F102C8T6, robot

devemin, 2019-07-13:
Hello everyone.

Could anyone tell me 'STM32F102C8T6' can use this bootloader ?

In japan, we could buy this robot for $200 ( about 20,000 JPN YEN) a few days ago.
(sold out now.  It may be in stock again.)
It is usually $1,500 (150,000 JPN YEN).

There are 25 servos, it's crazy.  The name is 'pre-maid AI'.
[https://twitter.com/kirurobo/status/1147714348346793984?s=20](url)

[https://www.amazon.co.jp/DMM-make-ROBOTS-RBHM0000000445731927-%E3%83%97%E3%83%AA%E3%83%A1%E3%82%A4%E3%83%89AI/dp/B01CNGMIG2](url)

[http://robots.dmm.com/robot/premaidai](url)

[http://izm-11.hatenablog.com/entry/2019/07/04/123333](url)

It's so exiciting now at twiiter. They are all hacking it now.
I don't use STM series MCU usually.
I think this is for STM32F103?

But this robot use STM32F102.
I want to hack this robot as arduino.

Please anyone tell me can I use this bootloader for STM32F102 ?

thanks.
devemin

devemin, 2019-07-18:
hello.

I tried it .
It work successfully.(maybe)

I don't know detail, but I can write this bootloader to STM32F102, and Arduino IDE's sample(LED Blink) is work.

And you can buy this robot (include serial servo x25, made by Kondo pro).
You can try it.
https://global.rakuten.com/en/store/grass-road/item/premaid-ai/

My work. (STM32F102 with Arduino tweet.)
https://twitter.com/devemin/status/1151753223352410114?s=20

thanks.

devemin, 2019-07-18:
ah, I had a mistake in my sentence.

"I tried it " means with my original STM32 board.
I haven't try it with Pre-maid AI built-in board.

but I will try after 1-2week!

thanks.

stevstrong, 2019-09-14:
I will close this because of missing relevancy to this core.
Just reopen it if you think this should not be marked as solved yet.

closed 2019-09-14 by stevstrong