rogerclarkmelbourne/Arduino_STM32 #648: Got error in "Adafruit_GFX_AS" because textsize not defined anymore in the "Adafruit_GFX"

HamidSaffari, 2019-07-15:
Hi,
Adafruit changed his library and the "textsize" variable changed to:
textsize_x,      ///< Desired magnification in X-axis of text to print()
textsize_y,      ///< Desired magnification in Y-axis of text to print()
so please update it in your Adafruit_GFX_AS lib.

victorpv, 2019-12-27:
Adafruit_GFX_AS was a port I did a long time ago based on someone else's library. It should be considered obsolete at this point.
If I remember right the main benefit was that allowed several font sizes, but I think the current Adafruit_GFX allows multiple fonts, including different sizes, doesn't it?
What are you trying to do with the _AS library that you couldn't with the standard one?

victorpv, 2019-12-27:
See https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/646

stevstrong, 2021-01-24:
Closing this as Adafruit_GFX_AS seems to be obsolete by now, as Victor mentioned.

closed 2021-01-24 by stevstrong