rogerclarkmelbourne/Arduino_STM32 #649: Broken link, on rogerclark.net page for Maple Mini bootloader - should point here?

Andy2No, 2019-07-24:
There is an article on rogerclark.net, about the Maple Mini bootloader, which gives a link to it on github, which no longer works.  It seems the folder structure was revised since that was written.

I couldn't see how to respond directly to that page, so I thought posting here might work.

https://www.rogerclark.net/improved-maple-bootloader-for-stm32/ says:

> All the code and sources have been pushed to GitHub , https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/usb_bootloader/STM32F1
> 
> including a binary for the Maple mini

Perhaps that page could be updated?

Serasidis, 2019-07-28:
Perhaps you are talking about this link
https://github.com/rogerclarkmelbourne/STM32duino-bootloader

Andy2No, 2019-07-28:
Yes, Serasidis, it looks like that's where it ought to point to now.

stevstrong, 2019-09-15:
Vassilis provided the right answer, so I consider this solved.
Maybe Roger will correct the link on his own site accordingly.

closed 2019-09-15 by stevstrong