rogerclarkmelbourne/Arduino_STM32 #651: Can't get Linux Mint 19.1 to recognize my board using the Stm32duino bootloader

mateusnazarioc, 2019-07-26:
Hello everyone!
I've acquired a Maple Mini clone from Baite, the BTE14-07A. They said it came with the bootloader to use with Arduino, but I can't get it to work with both OS that I said above.
Especially in Mint(my main OS), I tried all stages of the tutorial and the COM doesn't list the Serial.
Installed the DFU-Util v0.8, but it didn't work too...
Should I try to install the bootloader again?
This is the error I'm getting when trying to upload to the board via Arduino IDE.
[image]

Ah, and I'm also in the Perpetual Bootloader mode when trying to upload the sketch.

Thx for everything. Hope I can get this awesome board to work :)

ssieb, 2019-07-26:
Do you have a USB to 3.3V serial adapter?  I would recommend just using that to replace the bootloader with something you know is good.

mateusnazarioc, 2019-07-27:
> Do you have a USB to 3.3V serial adapter? I would recommend just using that to replace the bootloader with something you know is good.

Hey ssieb, thx for the reply! Gonna try this now....This is the "solution" that I thought too hahaha

mateusnazarioc, 2019-07-27:
So, I got this error...
[image]
Any ideas?

mateusnazarioc, 2019-07-27:
I've got it both on the Arduino IDE trying to upload via Serial and via stm32flash to upload the binary

mateusnazarioc, 2019-07-27:
So, after some tries, I was able to install the bootloader on the board. It starts with the 6 fast blinks, then the slow blinks, and then stops.
But I'm getting the same error in the Arduino IDE...
Any ideas?

Combinacijus, 2019-08-11:
I have same problem on Ubuntu 18.04

```
Waiting for /dev/ttyACM0 serial...Done
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
```

Interesting thing is that I have uploaded bootloader and on two STM32 and they are fully working on Window 10 Arduino IDE. I can upload code and open serial monitor everything ok.
But on Ubuntu can't upload code. Manage to open serial monitor tho.
You can also open serial in terminal by: `cat /dev/ttyACM0` at first it didn't work but found that after `sudo apt-get purge modemmanager` it started working (don't know why or consequences).

It's been a real pain to get to this point on Ubuntu, any ideas?

ssieb, 2019-08-11:
Because it shows up as an ACM device, it looks like a modem, which means that ModemManager will check to see if it is.  On Fedora, I just disabled ModemManger using systemctl.  If you aren't expecting to use a modem, then removing it should be fine as well.

stevstrong, 2019-09-15:
What is the status of this issue? It it solved? Can be closed?

stevstrong, 2019-09-16:
Possible duplicate of #585

rogerclarkmelbourne, 2020-02-09:
I'm going to close this, as there has been no activity from the OP for around 6 months

closed 2020-02-09 by rogerclarkmelbourne