rogerclarkmelbourne/Arduino_STM32 #652: Arduino STM32 Watchdog

Milad-Kt, 2019-07-28:
Hello
How to set up watchdog on Arduino_STM32?
and
How to set up watchdog on stm32 arduino when use FreeRTOS?
Please Help
with example or Code

thanks a lot

stevstrong, 2019-09-15:
I don't think such an example will ever be implemented, unless you provide a PR.

Milad-Kt, 2019-09-19:
Hello
thanks for response

Stm32f103RET6 have 2 Watchdog timer
[https://www.st.com/content/st_com/en/products/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus/stm32-mainstream-mcus/stm32f1-series/stm32f103/stm32f103re.html](url)
My specific question is
How do I enable this feature?
And use it?

my program is in Arduino and i am using Arduino_STM32 library

thanks a lot

stevstrong, 2019-09-19:
You have to implement the functionality described in RM0008 chapters 19.3 and 20.3.
Or you switch to the official core supported by STMicroelectronics and use [the example skecth](https://github.com/stm32duino/Arduino_Core_STM32/blob/1f6168462ce2ef0e88578773161956387d938c19/libraries/IWatchdog/examples/IWDG_Button/IWDG_Button.ino).

Milad-Kt, 2019-09-21:
thanks

It is possible that use register settings on Arduino with rogerclarkmelbourne/Arduino_STM32
on stm32??How?

my base code is rogerclarkmelbourne/Arduino_STM32 on arduino
so i need watchdog on arduino and with this library?

There really is no solution?

rogerclarkmelbourne, 2019-09-21:
The solution is you write the library.

Thats how all the other libraries got written that are in the repo.

People wrote the library because they needed it, and for the good of the community they submitted the library for inclusion to this repo

closed 2019-09-21 by rogerclarkmelbourne

ShaggyDog18, 2020-05-17:
support the above request:  could you please develop a library similar to [WatchDog Library](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/IWatchdog)
Thank you!