rogerclarkmelbourne/Arduino_STM32 #654: [README.md] Reference to stm32duino.com is outdated.

NB0174, 2019-08-06:
The site is dead now.

BlackBrix, 2019-08-06:
maybe it's due to webhosting problems described here:
https://www.rogerclark.net/blog-offline-due-to-server-quota-problems/

SuperUserNameMan, 2019-08-06:
also are the links to youtube videos ...

NB0174, 2019-08-06:
Possibly. 
In the meantime does anyone know his policy about reposting a package in a separate git for our firmware users ? We have two copies of STM32 Arduino but our users might well need a copy, for example when we roll out a firmware updates.

rogerclarkmelbourne, 2019-11-15:
I transferred the domain to ST a few months ago, as they told me they would be building a forum to replace the run I was running, but unfortunately that hasn't happened :-(

BlackBrix, 2019-11-15:
we are all here now https://www.stm32world.com

fpistm, 2019-11-15:
> I transferred the domain to ST a few months ago, as they told me they would be building a forum to replace the run I was running, but unfortunately that hasn't happened :-(

This will come.

rogerclarkmelbourne, 2020-02-09:
The new www.stm32duino.com has no been up any running for several months

closed 2020-02-09 by rogerclarkmelbourne