rogerclarkmelbourne/Arduino_STM32 #655: Serial numbers shifted if not using SERIAL_USB

saloid, 2019-08-06:
Hello, if I am using board without USB, all serial ports are shifted, so I should use Serial4 when i mean 3 port (in stm32CubeMX it's 3 port, and if i turn on `SERIAL_USB` it is 3 port too).
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/a3a56866505eeca4b49fc435700dcd4bd5540c8c/STM32F1/cores/maple/HardwareSerial.h#L184-L222
I think port numbers should be always the same (in old versions of this file they were same) and for example if `SERIAL_USB` disabled define `Serial` as `Serial1`

saloid, 2019-08-06:
Example: I am using STM32F103RCT6 mcu.
```
| Pins       | CubeMX | with USB | without USB |
|------------|--------|----------|-------------|
| PC10, PC11 | UART4  | Serial4  | Serial3     |
| PA2, PA3   | USART2 | Serial2  | Serial1     |
```

stevstrong, 2019-09-14:
Yes, you are right.
The solution would be to assign Serial x always to the corresponding U(S)ART interface, and leave Serial only for USB serial.
Can you prepare a PR which solved that?

saloid, 2019-09-17:
I have checked and found that Serial number assigns in every board individually.
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/e009a060eb58e4a7a619e07557ac7f476fc20915/STM32F1/variants/generic_stm32f103c/board.cpp#L127-L139
How better to do in this case? Change it in every board and add #warning that Serial has changed?

saloid, 2019-09-17:
And if no SERIAL_USB in this board, maybe better to define Serial as Serial1?

stevstrong, 2019-09-17:
I think no warning is necessary because when SERIAL_USB is not defined then
`#define Serial Serial1`
should do the trick.

saloid, 2019-09-17:
I mean if somebody already use shifted numbers (like me), and they will be no more shifted - in this case I'd prefer to be warned. And how about changing this - I need to change it in all boards, right?

stevstrong, 2019-09-17:
Yes, you need to be change that for all boards.

victorpv, 2019-12-27:
This could break compatibility with older sketches that expect that behavior. I remember it has been like that forever.

stevstrong, 2020-10-12:
PR merged.

closed 2020-10-12 by stevstrong