rogerclarkmelbourne/Arduino_STM32 #657: WS2812B_STM32_Libmaple doesnt work without delays

derbengale, 2019-08-12:
Hello, 
for my project I needed the LEDs to be as fast as possible why i chose the STM32 as the microcontroller of my choice.
Unfortunately it seems like the LEDs dont update unless there is a delay of at least 10ms between the updates.
So I cant use the library. 
Thanks for your work btw

ssieb, 2019-08-12:
You need to provide more details because that doesn't make sense.

derbengale, 2019-08-15:
Sure. I was using the STM32103F8 microcontroller and used the library exampe code from [here](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/WS2812B/examples/WS2812B_test/WS2812B_test.ino)
I just changed the loop to 

`void loop() 
{ 
  rainbowCycle(1);

}
`

and afterwards tried to remove the delay to see how fast it can update, but then the LED Stripe just doesnt light up anymore. When i add the delay again it starts working again but only if there is a delay of at least 10ms else the LEDs stay dark.
I couldn't figure out why that happens.

rogerclarkmelbourne, 2019-08-15:
The code uses DMA to send the data to the LEDs as a background process, using the SPI lines.

The code takes a certain amount of time to encode the new pattern before submitting the encoded data the DMR controller to send the data via SPI.

What's probably happening, is you are attempting to send another set of data before the first lot of data has finished being sent by the DMR controller.

Have you put a logic analyser on the data line to determine how much gap there is between the end of one transmission to the start of the next transmission.

Its been around 2 years since I wrote the library, and I can't remember the minimum delay, but it would depend on how long your code take to generate the new RGB values, e.g if you were generating a fractal sequence, then the processing time for the RGB generation would probably be quite long and you'd not need a delay at all.

enesbcs, 2019-08-15:
Did you try this library? Old but still works.
https://github.com/ANDnXOR/Adafruit_NeoPixel-ANDnXOR.git

derbengale, 2019-08-15:
I don't have a logic analyzer unfortunately. Maybe im gonna wish that for christmas :-)

Thank you enesbcs, I will try that!

closed 2019-08-15 by derbengale