rogerclarkmelbourne/Arduino_STM32 #660: pwmWrite don't work on STM32F407VGT6 Black Board

romanlife, 2019-08-27:
hi Roger, I am testing the Board on stm32f407vgt6 processor and cannot use pwmWrite(pin,value) function.When you build a sketch, it throws an error. Could you please explain to me how to initialize it? I'm using Arduino IDE 1.8.9, with your kernel

stevstrong, 2019-08-27:
The support for F4 family processors of this core is limited.
Instead, I would suggest you to use [my core](https://github.com/stevstrong/Arduino_STM32) which supports some generic F4 boards, or use the [official STM core](https://github.com/stm32duino/Arduino_Core_STM32) for any other official ST board.

romanlife, 2019-08-27:
Hi Steve, I will try it with your core? thanks

stevstrong, 2019-09-14:
I will close this here.
As explained, the F4 features are limited here.
Instead, for F4 support please refer in the future to [my repo](https://github.com/stevstrong/Arduino_STM32).

closed 2019-09-14 by stevstrong