rogerclarkmelbourne/Arduino_STM32 #669: The error when compile "Blink",in arduino_stm32

yifanshu02, 2019-09-02:
```
Arduino: 1.8.9 (Windows 10), Board: "Generic STM32F103V series, STM32F103VC, STLink, 72Mhz (Normal), Smallest (default)"

In file included from c:\users\administrator.sc-201906101001\appdata\local\arduino15\packages\stm32\tools\arm-none-eabi-gcc\8.2.1-1.7\arm-none-eabi\include\stdlib.h:19,
                 from c:\users\administrator.sc-201906101001\appdata\local\arduino15\packages\stm32\tools\arm-none-eabi-gcc\8.2.1-1.7\arm-none-eabi\include\c++\8.2.1\cstdlib:75,
                 from c:\users\administrator.sc-201906101001\appdata\local\arduino15\packages\stm32\tools\arm-none-eabi-gcc\8.2.1-1.7\arm-none-eabi\include\c++\8.2.1\stdlib.h:36,
                 from C:\arduino-1.8.9\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:41,
                 from C:\arduino-1.8.9\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30,
                 from sketch\Blink.ino.cpp:1:
C:\arduino-1.8.9\hardware\Arduino_STM32\STM32F1\system/libmaple/stm32f1/include/series/gpio.h:489:15: error: duplicate 'inline'
 static inline __always_inline void afio_exti_select(exti_num exti, exti_cfg port) {
               ^~~~~~~~~~~~~~~
C:\arduino-1.8.9\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/nvic.h:112:15: error: duplicate 'inline'
 static inline __always_inline void nvic_globalirq_enable() {
               ^~~~~~~~~~~~~~~
C:\arduino-1.8.9\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/nvic.h:119:15: error: duplicate 'inline'
 static inline __always_inline void nvic_globalirq_disable() {
               ^~~~~~~~~~~~~~~
C:\arduino-1.8.9\hardware\Arduino_STM32\STM32F1\cores\maple/ext_interrupts.h:109:15: error: duplicate 'inline'
 static inline __always_inline void interrupts() {
               ^~~~~~~~~~~~~~~
C:\arduino-1.8.9\hardware\Arduino_STM32\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
 static inline __always_inline void noInterrupts() {
               ^~~~~~~~~~~~~~~
exit status 1
Error compiling for board Generic STM32F103V series.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
how to solve this problem?

rogerclarkmelbourne, 2019-09-02:
I did a test build on a blank project using Arduino 1.8.9 and I didnt get any errors

I'm not sure why your setup is using GCC 8.2.1

What "Board" did you install to get the ARM compiler

yifanshu02, 2019-09-02:
thanks for your reply,sir.
the error caused by GCC 8.2.1 ?
[image]

yifanshu02, 2019-09-02:
sir,i solve it 
i want that i reinstalled the WIN system,
i ignored that:
`Run the IDE, and on the Tools menu, select the Boards manager, and install the Arduino SAM boards (Cortex-M3) from the list of available boards. You must do this step, it installs the arm-none-eabi-g++ toolchain!`
sorry for the embarrassing mistake:D

closed 2019-09-02 by yifanshu02

rogerclarkmelbourne, 2019-09-02:
OK.

No worries...