rogerclarkmelbourne/Arduino_STM32 #670: Bootloader Overwritten when Blink uploaded

zpowellman, 2019-09-03:
Hi All,
I uploaded and burned the bootloader in a blue pill board, reset Boot 0 jumper and connected to USB cable.  Device appeared as Maple Com30.  
On the Arduino IDE I selected Generic STM32F1 Series,  BluePill F103C8  and upload method as Maple DFU Bootloader 2.0  I then uploaded the 'Blink' program.  Blink program works well, but the comport disappeared and it looks like the bootloader is gone.  The device now shows as "Unknown USB Device" in Device Manager.  I then have to reprogram the bootloader again. 
Uploading anything wipes out the bootloader.  Where did I go wrong?  
Thanks,
Dan Powell

rogerclarkmelbourne, 2019-09-03:
Those options don't apply to this Arduino Core. 

Please post the issue to the appropriate Github repo (note none of the repos in my account have those options). I suspect you may be using the official STM32 core https://github.com/stm32duino

closed 2019-09-03 by rogerclarkmelbourne

zpowellman, 2019-09-03:
Thanks for the response.

zpowellman, 2019-09-03:
I didn't know there were different systems. Are you affiliated with the stm32duino group or is yours a totally different distribution?  I think I used yours a couple years ago.  Will both systems coexist in the Arduino IDE?  
Again, thanks for the clarification. 
Dan Powell

rogerclarkmelbourne, 2019-09-03:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Credits-and-history