rogerclarkmelbourne/Arduino_STM32 #671: Increase Flash Memory (Program Memory)

nikjayswal, 2019-09-05:
How to increase Flash Memory (program memory) using SPI External Flash Memory like W25Q32 i am using stm32f103c8t6. my program is running out of memory i need to increase memory using external SPI memory. I had search on internet but doesn't find any tutorial/examples how to increase memory using external spi.

rogerclarkmelbourne, 2019-09-05:
You can't use external memory as part of your program variables etc. Its not supported by the MCU or the compiler

closed 2019-09-05 by rogerclarkmelbourne

nikjayswal, 2019-09-05:
1) I found this thread From this link  https://arduino.stackexchange.com/q/19489

2) Is dual boot possible to boot from SD Card to run Arduino sketch instead of internal Flash Storage ?