rogerclarkmelbourne/Arduino_STM32 #680: STM32F103CB SPI _currentSettings is a NULL pointer

BSG-SalvoL, 2019-09-19:
For some reason i'm experiencing this issue due to the parameter passed to SPIClass constructor.
Use u8g2 library was impossible, i've to add a reset function passing the SPI number (like the constructor) to keep the SPI working.
Here the reset Function.
`void SPIClass::resetSettings(uint8_t num) {
    _currentSetting = new SPISettings(SPI_CLOCK_DIV128, MSBFIRST, SPI_MODE3, DATA_SIZE_8BIT);
    switch(num)
    {
        case 1:
            _currentSetting->spi_d = SPI1;
            _currentSetting->clockDivider = determine_baud_rate(_currentSetting->spi_d, _currentSetting->clock);
            _currentSetting->spiDmaDev = DMA1;
            _currentSetting->spiTxDmaChannel = DMA_CH3;
            _currentSetting->spiRxDmaChannel = DMA_CH2;
        break;

#if BOARD_NR_SPI >= 2
        case 2:
            _currentSetting->spi_d = SPI2;
            _currentSetting->clockDivider = determine_baud_rate(_currentSetting->spi_d, _currentSetting->clock);
            _currentSetting->spiDmaDev = DMA1;
            _currentSetting->spiTxDmaChannel = DMA_CH5;
            _currentSetting->spiRxDmaChannel = DMA_CH4;
        break;
#endif

#if BOARD_NR_SPI >= 3
        case 3:
            _currentSetting->spi_d = SPI3;
            _currentSetting->clockDivider = determine_baud_rate(_currentSetting->spi_d, _currentSetting->clock);
            _currentSetting->spiDmaDev = DMA2;
            _currentSetting->spiTxDmaChannel = DMA_CH2;
            _currentSetting->spiRxDmaChannel = DMA_CH1;
#endif
    }
}
`

U8g2 library uses SPI_INTERFACES_COUNT (which is undefined) instead of BOARD_NR_SPI.

stevstrong2, 2019-09-19:
Without knowing your application I cannot tell you what you did wrong.
Normally, the _currentSetting is always non-null, it will be initialized with SPI.begin() or SPI.beginTransaction(). You have to call one of these two functions before being able to use the SPI API functions.

BSG-SalvoL, 2019-09-19:
The SPI library is directly used by u8g2 lib

stevstrong2, 2019-09-20:
Please provide an example sketch which demonstrated the issue.
As I told, you have to call in your setup one of the SPI begin functions.

BSG-SalvoL, 2019-09-20:
`#include "Arduino.h"
#include <U8g2lib.h>
#include <U8x8lib.h>

U8G2_ST7920_128X64_F_2ND_HW_SPI *u8g2;
uint8_t x, y;
static const uint8_t LCD_CS = PB12;

void setup() {
  u8g2 = new U8G2_ST7920_128X64_F_2ND_HW_SPI(U8G2_R0, LCD_CS, U8X8_PIN_NONE);
  u8g2->begin();
  u8g2->setFontRefHeightExtendedText();
  u8g2->setDrawColor(1);
  u8g2->setFontPosTop();
  u8g2->setFontDirection(0);
  x = y = 0;
}

void loop() {
  u8g2->clearBuffer();
  u8g2->setFont(u8g2_font_5x8_tf);
  u8g2->drawStr(x, y, "THIS IS A TEST");
  u8g2->sendBuffer();

  x++;
  if(x > 128) {
    x = 0; 
    y++;
    if(y > 64)
      y = 0;
  }
}`

This is not working for me:
In U8G2 for the second HW SPI is checked and enabled if SPI_INTERFACES_COUNT > 1; which is undefined for this STM32 in boards.h (in which is defined BOARD_NR_SPI).
Defining SPI_INTERFACES_COUNT 2 -> `#define SPI_INTERFACES_COUNT 2`;
Now U8G2 uses SPI1 like an SPIClass but it isn't. Then i've to define an SPIClass object and modify the calls. Done this when i run the sketch the board freeze due to NULL pointer in _currentSettings.
The first time U8G2 call the begin function, so it's use of the SPI interface is correct. (It work with a many boards in HW SPI)

stevstrong, 2019-09-20:
I cannot see that U8G2 library has support for STM32F103x8 controller family.

This having said, you cannot expect that this core fulfills some special needs of a particular library.
You should ask for support from the owner of this lib to implement support for this controller and core.
As far as I could see it has some files related to STM32l011 and l031, but using the official STM core.

Nevertheless, if you want to use SPI of this core, you have to call one of the begin functions mentioned in a reply above.
Additonally, if you use SPI port 2, then you have to call
`SPI.setModule(2);`
before calling any SPI function.

As long as you do not conform to these requirements, your issue is actually classified as not relevant.

BSG-SalvoL, 2019-09-20:
Because i've resolved my problem using that function, i can say that debugging the parameter passed in constructor (which in my case is 2) inside the constructor is seen as zero. But only there, in fact the resetFunction that i've created do the same thing and it works properly. Maybe can it be a problem of my board? Nobody is experiencing this? 
I've tried many times to reinstall my software (Arduino IDE and PlatformIO) and the libraries with always the same result.

stevstrong2, 2019-09-20:
As long as the library does not support the STM32F1 series, you will get the same problem no matter how many times you reinstall the IDE.
Ask the library owner for support.

dewhisna, 2019-09-21:
FWIW, the U8G2 library in general **does** work with the STM32F1.  Or at least the 2.24.3 one from this bundle: [http://downloads.arduino.cc/libraries/github.com/olikraus/U8g2-2.24.3.zip](http://downloads.arduino.cc/libraries/github.com/olikraus/U8g2-2.24.3.zip) does, as I'm using it without issue on a STM32F103.  However, I've only ever tried it using the I2C interface (not the SPI interface as @BSG-SalvoL is), but I don't see why it wouldn't work with SPI.

@BSG-SalvoL I think what is happening is there's a collision of names.  Arduino_STM32 has a `SPI1` defined (which is the name normally used in the SPI library for the second SPI port), but it's of type `spi_dev *`, declared in `system/libmaple/stm32f2/include/series/spi.h`  and instantiated in `cores/maple/libmaple/spi_f1.c`.

But, the U8g2 library is expecting SPI1 to be a `extern SPIClass SPI1;` like the non-STM32 SPI library uses.  The SPI library in this Arduino_STM32 framework only has a single `SPIClass SPI` object defined and it's initially set to the first SPI port.  The SPI library doesn't create `SPIClass SPI1` even when the boards file is set for two SPI modules (most likely because of the name collosion).

I do see one way for you to do it...  Change your instantiation of U8g2 to use the first SPI port so that it uses the `SPIClass SPI` object, and call the `setModule` function on the `SPI` object to switch it from the first SPI port to the second SPI port before calling the `u8g2->begin` by doing this:

```C++
#include "Arduino.h"
#include <SPI.h>      // Need to include SPI header to call setModule
#include <U8g2lib.h>
#include <U8x8lib.h>

U8G2_ST7920_128X64_F_HW_SPI *u8g2;    // Use the main HW_SPI, not 2ND_HW_SPI
uint8_t x, y;
static const uint8_t LCD_CS = PB12;

void setup() {
  SPI.setModule(2);   // <<<< Switch to 2nd SPI
  u8g2 = new U8G2_ST7920_128X64_F_HW_SPI(U8G2_R0, LCD_CS, U8X8_PIN_NONE);     // Instantiate the HW_SPI object instead of 2ND_HW_SPI
  u8g2->begin();
  u8g2->setFontRefHeightExtendedText();
  u8g2->setDrawColor(1);
  u8g2->setFontPosTop();
  u8g2->setFontDirection(0);
  x = y = 0;
}
```

stevstrong, 2019-12-26:
Any update on this?

BSG-SalvoL, 2020-01-02:
Hi, i've no updates, with Little corrections which i made for myself my sketches work with u8g2 lib in HW mode on the 2nd SPI interface. The suggestion proposed by @dewhisna Is interesting but i've other components in the First SPI Port, this mean that i should switch continously the SPI module and i don't know if the pointer problem Is resolved in this way. Maybe with the help of @olikraus which knows about the problem too you can reach a common fix for the full support of u8g2 library for f103 boards.

stevstrong, 2020-01-02:
I don't understand one thing. Why you declare a pointer to the u8g8 lib, and why not use the instance of it instead?
```
#include "Arduino.h"
#include <U8g2lib.h>
#include <U8x8lib.h>

#define LCD_CS   PB12

//U8G2_ST7920_128X64_F_2ND_HW_SPI *u8g2; // instead use:
U8G2_ST7920_128X64_F_2ND_HW_SPI   u8g2(U8G2_R0, LCD_CS, U8X8_PIN_NONE);

uint8_t x, y;

void setup() {
  // you do not needed this line anymore, the instance is already defined before setup
  //u8g2 = new U8G2_ST7920_128X64_F_2ND_HW_SPI(U8G2_R0, LCD_CS, U8X8_PIN_NONE);
  u8g2.begin();
  u8g2.setFontRefHeightExtendedText();
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
  x = y = 0;
}

void loop() {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_5x8_tf);
  u8g2.drawStr(x, y, "THIS IS A TEST");
  u8g2.sendBuffer();

  x++;
  if(x > 128) {
    x = 0; 
    y++;
    if(y > 64)
      y = 0;
  }
}
```
Please try this version and let us know if it works or not.

olikraus, 2020-01-02:
> Maybe with the help of @olikraus which knows about the problem too you can reach a common fix for the full support of u8g2 library for f103 boards.

As a lib auther it seems to be very difficult to detect and use the second SPI or Wire interface.
Obviously I would prefer a common way across all boards. 

The original Arduino libs use Wire1:
https://github.com/arduino/ArduinoCore-samd/blob/5dced38e81e6001a4b6e237e3462bc8db962127a/libraries/Wire/Wire.h#L91

and SPI1:
https://github.com/arduino/ArduinoCore-samd/blob/5dced38e81e6001a4b6e237e3462bc8db962127a/libraries/SPI/SPI.h#L162

Detection is possible via `WIRE_INTERFACES_COUNT` and `SPI_INTERFACES_COUNT`.

stevstrong, 2020-01-02:
As @dewhisna already suggested, using SPI1 should work out of the box.
```
//U8G2_ST7920_128X64_F_2ND_HW_SPI *u8g2; // instead use:
U8G2_ST7920_128X64_F_HW_SPI   u8g2(U8G2_R0, LCD_CS, U8X8_PIN_NONE); // use default SPI interface

```

BSG-SalvoL, 2020-01-02:
> I don't understand one thing. Why you declare a pointer to the u8g8 lib, and why not use the instance of it instead?
> 
> ```
> #include "Arduino.h"
> #include <U8g2lib.h>
> #include <U8x8lib.h>
> 
> #define LCD_CS   PB12
> 
> //U8G2_ST7920_128X64_F_2ND_HW_SPI *u8g2; // instead use:
> U8G2_ST7920_128X64_F_2ND_HW_SPI   u8g2(U8G2_R0, LCD_CS, U8X8_PIN_NONE);
> 
> uint8_t x, y;
> 
> void setup() {
>   // you do not needed this line anymore, the instance is already defined before setup
>   //u8g2 = new U8G2_ST7920_128X64_F_2ND_HW_SPI(U8G2_R0, LCD_CS, U8X8_PIN_NONE);
>   // you need this line to switch to second SPI port
>   SPI.setModule(2);
>   // now you can use SPI2
>   u8g2.begin();
>   u8g2.setFontRefHeightExtendedText();
>   u8g2.setDrawColor(1);
>   u8g2.setFontPosTop();
>   u8g2.setFontDirection(0);
>   x = y = 0;
> }
> 
> void loop() {
>   u8g2.clearBuffer();
>   u8g2.setFont(u8g2_font_5x8_tf);
>   u8g2.drawStr(x, y, "THIS IS A TEST");
>   u8g2.sendBuffer();
> 
>   x++;
>   if(x > 128) {
>     x = 0; 
>     y++;
>     if(y > 64)
>       y = 0;
>   }
> }
> ```
I think that making the example for you from my code i forget to remove that line, ignore It.

BSG-SalvoL, 2020-01-02:
> As @dewhisna already suggested, using SPI1 should work out of the box.
> 
> ```
> //U8G2_ST7920_128X64_F_2ND_HW_SPI *u8g2; // instead use:
> U8G2_ST7920_128X64_F_HW_SPI   u8g2(U8G2_R0, LCD_CS, U8X8_PIN_NONE); // use default SPI interface
> ```

As confirmed by me and @dewhisna the SPI1 variable Is not of type SPIClass, so it's impossibile to use, while as suggested by @olikraus too, SPI_COUNT_INTERFACES is used to detect the Number of SPI, which is undefined in boards.h but standard in the Arduino library. 

I've made modifications in both u8g2 and in this library, to keep the code working. This Is why i mention olikraus in this post.

stevstrong, 2020-01-02:
AFAIK, SPI_COUNT_INTERFACES is used in U8G2 only to make use of SPI2 or other, the default SPI class will use the spi 1 interface. So even without this define the code should work for spi1.

SPI1 is a pointer to the **core intern** spi1 device.
SPI is the default instance of a SPI class defined to use the spi1 device, but you can switch the scope of SPI to use spi 2 device as I pointed out above. This way you must not touch the U8G2 lib and still you can use spi 2 port, as you wish.

dewhisna, 2020-01-02:
The real issue with the U8G2 library is that it's trying to be too helpful by having macros defined for every possible combination of hardware options.  If it would instead just let you pass the name of the `SPI` variable to use as an argument to the macro, you could define your `SPI2` object under whatever name you wanted to call it, rather than U8G2 being hardcoded for specific names, and there would be no issue.

@BSG-SalvoL -- you mentioned needing additional libraries running on SPI1 and didn't want to keep switching this SPI object between the two ports (which is reasonable).  However, my question is do those other libraries also hardcode the `SPI` variable name like U8G2 does??  If not, you can simply declare your own `SPI1` instance under whatever name you wanted to (such as `mySPI1`) and pass it to those other libraries and keep both ports active under two different objects.

Another option, though a lot more difficult to pull off successfully and may have build-system dependencies, would be to declare a second SPI instance for SPI2 and trick the U8G2 library to use it as `SPI` instead of the name declared in the core.  It would require some linking shenanigans, which is why I didn't originally suggest it, but I think it could be done.

The best bet would be to either change U8G2 to allow you to simply pass the SPI variable name to its macros or do that for your other libraries if they are also relying on the hardcoded `SPI` name.  I would be surprised if they were all hardcoded, as the better written libraries generally allow the user to pass whatever variable name they wish either as a reference argument to a class or as an argument to a macro instead of hardcoding them.

olikraus, 2020-01-02:
> If it would instead just let you pass the name of the SPI variable to use as an argument to the macro

Yes, indeed, I think it would be a better approach to pass the desired SPI / Wire object to the the u8g2 constructor.

closed 2020-02-06 by rogerclarkmelbourne