rogerclarkmelbourne/Arduino_STM32 #683: The code below works on stm32f103c8t6 blue pill:

rtek1000, 2019-09-23:
The code below works on stm32f103c8t6 blue pill:

```
void setup() {
  // Source: https://community.platformio.org/t/pio-arduino-stm32f103cbt6-remap/6786/5
  afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY); // PB3 free
  afio_remap(AFIO_REMAP_SPI1);

  gpio_set_mode (GPIOB, 3, GPIO_AF_OUTPUT_PP);
  gpio_set_mode (GPIOB, 4, GPIO_INPUT_FLOATING);
  gpio_set_mode (GPIOB, 5, GPIO_AF_OUTPUT_PP);

// put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Example:

```
/**
    SPI_1 and SPI_2 port example code

    Description:
    This sketch sends one byte with value 0x55 over the SPI_1 or SPI_2 port.
    The received byte (the answer from the SPI slave device) is stored to the <data> variable.

    The sketch as it is, works with SPI_1 port. For using the SPI_2 port, just
    un-comment all the nessesary code lines marked with <SPI_2> word.

    Created on 10 Jun 2015 by Vassilis Serasidis
    email:  avrsite@yahoo.gr

    Using the first SPI port (SPI_1)
    SS    <-->  PA4 <-->  BOARD_SPI1_NSS_PIN
    SCK   <-->  PA5 <-->  BOARD_SPI1_SCK_PIN
    MISO  <-->  PA6 <-->  BOARD_SPI1_MISO_PIN
    MOSI  <-->  PA7 <-->  BOARD_SPI1_MOSI_PIN

    Using the first SPI port (SPI_1) REMAPPED
    SS    <-->  PA15 <-->  BOARD_SPI1_NSS_PIN
    SCK   <-->  PB3 <-->  BOARD_SPI1_SCK_PIN
    MISO  <-->  PB4 <-->  BOARD_SPI1_MISO_PIN
    MOSI  <-->  PB5 <-->  BOARD_SPI1_MOSI_PIN

    Using the second SPI port (SPI_2)
    SS    <-->  PB12 <-->  BOARD_SPI2_NSS_PIN
    SCK   <-->  PB13 <-->  BOARD_SPI2_SCK_PIN
    MISO  <-->  PB14 <-->  BOARD_SPI2_MISO_PIN
    MOSI  <-->  PB15 <-->  BOARD_SPI2_MOSI_PIN
*/

#include <SPI.h>

#define SPI1_NSS_PIN PA15 //PA4    //SPI_1 Chip Select pin is PA4. You can change it to the STM32 pin you want.
#define SPI2_NSS_PIN PB12   //SPI_2 Chip Select pin is PB12. You can change it to the STM32 pin you want.

SPIClass SPI_2(2); //Create an instance of the SPI Class called SPI_2 that uses the 2nd SPI Port
byte data;

void setup() {
  // Remap SPI 1
  // Source: https://community.platformio.org/t/pio-arduino-stm32f103cbt6-remap/6786/5
  afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY); // PB3 free
  afio_remap(AFIO_REMAP_SPI1);

  gpio_set_mode (GPIOB, 3, GPIO_AF_OUTPUT_PP);
  gpio_set_mode (GPIOB, 4, GPIO_INPUT_FLOATING);
  gpio_set_mode (GPIOB, 5, GPIO_AF_OUTPUT_PP);
  
  // Setup SPI 1
  SPI.begin(); //Initialize the SPI_1 port.
  SPI.setBitOrder(MSBFIRST); // Set the SPI_1 bit order
  SPI.setDataMode(SPI_MODE0); //Set the  SPI_2 data mode 0
  SPI.setClockDivider(SPI_CLOCK_DIV16);      // Slow speed (72 / 16 = 4.5 MHz SPI_1 speed)
  pinMode(SPI1_NSS_PIN, OUTPUT);

  // Setup SPI 2
  SPI_2.begin(); //Initialize the SPI_2 port.
  SPI_2.setBitOrder(MSBFIRST); // Set the SPI_2 bit order
  SPI_2.setDataMode(SPI_MODE0); //Set the  SPI_2 data mode 0
  SPI_2.setClockDivider(SPI_CLOCK_DIV16);  // Use a different speed to SPI 1
  pinMode(SPI2_NSS_PIN, OUTPUT);

}

void loop() {

  sendSPI();
  sendSPI2();

  delayMicroseconds(10);    //Delay 10 micro seconds.
}

void sendSPI()
{
  digitalWrite(SPI1_NSS_PIN, LOW); // manually take CSN low for SPI_1 transmission
  data = SPI.transfer(0x55); //Send the HEX data 0x55 over SPI-1 port and store the received byte to the <data> variable.
  digitalWrite(SPI1_NSS_PIN, HIGH); // manually take CSN high between spi transmissions
}

void sendSPI2()
{
  digitalWrite(SPI2_NSS_PIN, LOW); // manually take CSN low for SPI_2 transmission
  data = SPI_2.transfer(0x55); //Send the HEX data 0x55 over SPI-2 port and store the received byte to the <data> variable.
  digitalWrite(SPI2_NSS_PIN, HIGH); // manually take CSN high between spi transmissions
}
```

_Originally posted by @rtek1000 in https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/536#issuecomment-533972563_

rtek1000, 2019-09-23:
Ops!

closed 2019-09-23 by rtek1000

Atlas85, 2020-08-10:
Hi,
The SPI1 AF works propperly, but I can't use PA5-PA7 anymore as GPIO. What can I do ?

`//******************************************* Pin Definitions ***************************************//
#define Relais_1    PA0     // Relais_1 Output
#define Relais_2    PA1     // Relais_2 Output
#define Relais_3    PA2     // Relais_3 Output
#define Relais_4    PA3     // Relais_4 Output
#define Relais_5    PA4     // Relais_5 Output
#define Relais_6    PA5     // Relais_6 Output
#define Relais_7    PA6     // Relais_7 Output
#define Relais_8    PA7     // Relais_8 Output

#define LED_LINK    PB0     // Link Led green
#define LED_ERROR   PB1     // Link Led red
#define SCK         PB3     // MCP2515  Clock Source
#define MISO        PB4     // MCP2515  Data to µC
#define MOSI        PB5     // MCP2515  Data from µC
#define MCP_CS      PB6     // MCP2515 /CS-Signal
#define MCP_INT     PB7     // MCP2515 INT-Signal

#define OB_LED_1    PB12    // Onboard LED 1
#define OB_LED_2    PB13    // Onboard LED 2

#define OB_LED      PC13    // Onboard LED 0

//***********************Debug_MSG's*******************************************************
//#define DEBUG_RELAIS
//#define DEBUG_CAN_TX
//#define DEBUG_CAN_init
//#define DEBUG_CAN_RX
//#define DEBUG_STATUS
//#define DEBUG_VALUES
//#define DEBUG_Setup

#include <SPI.h>
#include <mcp_can.h>
MCP_CAN CAN0(MCP_CS);
...
//******************************************** INIT PINMODES *********************************************//
void PINMODES() {

  pinMode(Relais_1,   OUTPUT);          // Relais_1 OutputDriver
  pinMode(Relais_2,   OUTPUT);          // Relais_2 OutputDriver
  pinMode(Relais_3,   OUTPUT);          // Relais_3 OutputDriver
  pinMode(Relais_4,   OUTPUT);          // Relais_4 OutputDriver
  pinMode(Relais_5,   OUTPUT);          // Relais_5 OutputDriver
  pinMode(Relais_6,   OUTPUT);          // Relais_6 OutputDriver
  pinMode(Relais_7,   OUTPUT);          // Relais_7 OutputDriver
  pinMode(Relais_8,   OUTPUT);          // Relais_8 OutputDriver
  pinMode(LED_LINK,   OUTPUT);          // Link LED
  pinMode(LED_ERROR,  OUTPUT);          // Error LED
  pinMode(SCK,        OUTPUT);          // SPI for MCP2515 and ADC
  pinMode(MISO,       INPUT_PULLUP);    // SPI for MCP2515 and ADC
  pinMode(MOSI,       OUTPUT);          // SPI for MCP2515 and ADC
  pinMode(MCP_INT,    INPUT_PULLUP);    // CAN with MCP2515 INT-Signal
  pinMode(MCP_CS,     OUTPUT );         // CAN with MCP2515 /CS-Signal
  pinMode(OB_LED,     OUTPUT);          // OnBoard LED active LOW!
  pinMode(OB_LED_1,   OUTPUT);          // OnBoard LED active LOW!
  pinMode(OB_LED_2,   OUTPUT);          // OnBoard LED active LOW!

    // Source: https://community.platformio.org/t/pio-arduino-stm32f103cbt6-remap/6786/5
  afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY); // PB3 free
  afio_remap(AFIO_REMAP_SPI1);

  gpio_set_mode (GPIOB, 3, GPIO_AF_OUTPUT_PP);
  gpio_set_mode (GPIOB, 4, GPIO_INPUT_FLOATING);
  gpio_set_mode (GPIOB, 5, GPIO_AF_OUTPUT_PP);

/* set pin mode */
/*
  gpio_set_mode (GPIOA, 5, GPIO_OUTPUT_PP);
  gpio_set_mode (GPIOA, 6, GPIO_OUTPUT_PP);
  gpio_set_mode (GPIOA, 7, GPIO_OUTPUT_PP);
*/

}

...

//********************************************** S E T U P ***********************************************//
void setup() {
  Serial1.begin(115200);

#ifdef DEBUG_Setup
  Serial1.println("UART1 init");
#endif

  PINMODES();

#ifdef DEBUG_Setup
  Serial1.println("I/O Pins Set");
#endif

  initMCP_CAN();

#ifdef DEBUG_Setup
  Serial1.println("Setup CAN done");
#endif
}
`
I tryed it with the out commented lines, resort some definitions, but still there comes nothing from/to PA5-7...
Tryed to make the AF Pin definition to the main SPI1 definition and deleted the AF Stuff, but then nothing worked anymore...
Would be nice, if there is a fix for this issue.
Thanks in advance 
Best Regards Denis Koellner

stevstrong, 2020-08-12:
You have to set the pin modes after the `SPI.begin()` is called, which is done most probably in `initMCP_CAN()`.

Atlas85, 2020-08-12:
Hi, for the next boards i will try it. For this time i have wirewrapped to some other free ports. It is not so good looking, but works... 
Thanks for the hint!!!
Best regards