rogerclarkmelbourne/Arduino_STM32 #685: STM32F407VGT don´t print right

arnold-b, 2019-09-24:
Hello, I´m using STM32407VGT generic Board. (TQFP100)

TX = PA9 (pin 68)
RX = PA10 (pin 69)

Serial.begin(115200);
  Serial.println("ILI9488 Test!");

Result: ⸮$3(⸮⸮H

On every try the same result. May speed Problem or ?

arnold-b, 2019-09-24:
i guess i found my problem. 

My board is using 12Mhz Crystal. But till now i didnt find where i can change the setting

rogerclarkmelbourne, 2019-09-24:
You will need to make youself a custom variant

Copy the whole solder and then look at the PLL settings and the F_CPU settings.

If you look at the GD32 variant I think it used a 12Mhz Crystal, so that would be a good starting point

closed 2019-09-24 by rogerclarkmelbourne

arnold-b, 2019-09-24:
I changed the Crystal to 8Mhz. Now it works.
But i would like to change back to 12Mhz, where can i change the Arduino settings for that ?
Thanks

rogerclarkmelbourne, 2019-09-24:
There are no "Arduino settings" for this.

Its intrinsic to the board "Variant"

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/variants

Look carefully at this variant

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/variants/generic_gd32f103c

Including

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/variants/generic_gd32f103c/wirish/boards_setup.cpp#L54-L56

stevstrong, 2019-09-24:
You could eventually have a look in my repo where one can define the crystal frequency:
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/libmaple/rccF4.c#L366-L372

arnold-b, 2019-09-24:
Hello, 
thanks together. for help