rogerclarkmelbourne/Arduino_STM32 #686: SPI problem

arnold-b, 2019-09-24:
Hello, 

my SPI dont work on STM32F103VCT6 and STM32F407VGT
on both i try

ILI9488(TFT_CS, TFT_DC,PA7,PA5,TFT_RST,PA6) or ILI9488 tft = ILI9488(TFT_CS, TFT_DC,TFT_RST);
#define TFT_CS         PB2  //PD15//PA1
#define TFT_DC         PB1
#define TFT_LED        PB0
#define TFT_RST        -1//PB4

if i just make IO test the PINS are fine. But on spi no clock and no mosi.
What is my Problem ?

stevstrong, 2019-09-24:
Please post the example sketch you are testing.
Where did you get the ILI9488 library from? Post a link to that.

arnold-b, 2019-09-24:
Hey, 
sorry i forgot that this is not from your samples. 
https://github.com/jaretburkett/ILI9488

But if i just try a simple SPI like this: 

void setup() {
  Serial.begin(115200);
  Serial.println("ILI9488 Test!");
  

 SPI.begin(); //Initialize the SPI_1 port.

}
void loop(void) {
Serial.println("Loop IO");

delayMicroseconds(1000); 
SPI.transfer(0xb4);
}

I don´t see anything.  No clock, nothing 
Do i have to define the PINS ? 
Sorry for my stupid questions. I´m new with STM. Till now i used ESP and Microchip (PIC)

D05/PA5  | 30 | SPI1_SCK   | OTG_HS_ULPI_CK | TIM2_CH1_ETR     | TIM8_CH1N                                                       | ADC12_IN5 / DAC_OUT2
    {&GPIOA,    NULL, 1,    6, &ADC1}, // D06/PA6  | 31 | SPI1_MISO  | TIM8_BKIN      | TIM13_CH1        | DCMI_PIXCLK    | TIM3_CH1       | TIM1_BKIN                     | ADC12_IN6
    {&GPIOA,    NULL, 0,    7, &ADC1}, // D07/PA7  | 32 | SPI1_MOSI  |

Board Generic STM32F407V

stevstrong, 2019-09-24:
The default SPI port for the generic F407 boards is port 3.
If you want to use port 1, then you have to call
`SPI.setModule(1);`
before `SPI.begin()`, or before any other library function which uses SPI.

arnold-b, 2019-09-25:
Hey, 
i give up. Waist already 3 Days only for a simple, stable SPI comunication. 
This also don´t have a good result.  
Why its not just simple. that user can define the SPI ports and its running. 
Even there are limitations from datasheet. thats easy to read. But the restrictions in lib´s are hard to find

arnold-b, 2019-09-25:
I will try one more time with STM32103 if this also don´t work. i need to decide to use Cube or just give up this PRODUCT. Learning time from Cube is to long. not much samples

stevstrong, 2019-09-25:
You have to control the CS pin by yourself. This is missing from your example.

arnold-b, 2019-09-25:
Hi. Thats my last sample. I expected that i can See on Logicanalyser 0xb4. But i  don´t got that.

void loop(void) {
Serial.println("Loop IO");

delayMicroseconds(100); 
digitalWrite(PB4, HIGH);
SPI.transfer(0xb4);
digitalWrite(PB4, LOW);
}

arnold-b, 2019-09-25:
and finaly i need to use STM32F103. 407 is to expensive for this project. I´m just soldering a small testing board. Got the Chips just now.

stevstrong, 2019-09-25:
You should post a complete example, not just a part of it and specify which target you use and which pins are you measuring.

arnold-b, 2019-09-25:
Hey, now i tested STM32F103VCT6 Chip. No special board. 
SPI1 and SPI2 is working as expected. 
Only SPI3 is not working. Clock is missing on PIN 89.

#include <SPI.h>

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin PB1 as an output.
  pinMode(PB1, OUTPUT);
  Serial.begin(115200);
  Serial.println("ILI9488 Test!");
  Serial1.begin(115200);
  Serial1.println("ILI9488 Test!");
 Serial2.begin(115200);
  Serial2.println("ILI9488 Test!"); 
  Serial3.begin(115200);
  Serial3.println("ILI9488 Test!");
  SPI.setModule(3);
  SPI.begin();
}

// the loop function runs over and over again forever
void loop() {
 
  Serial.println("Loop IO");
Serial.println("HWSPI0");
 Serial1.println("HWSPI1");
 Serial2.println("HWSPI2");
 Serial3.println("HWSPI3");
  digitalWrite(PB1, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(200);     
  SPI.transfer(0xb4);// wait for a second
  digitalWrite(PB1, LOW);    // turn the LED off by making the voltage LOW
  
  delay(200);              // wait for a second
}

[image]

dewhisna, 2019-09-25:
I believe that pin normally defaults to JTAG operation. See the [ST Datasheets](https://www.st.com/en/microcontrollers-microprocessors/stm32f103vc.html).
That means you need to disable the JTAG interface either completely or at least switch it to the SW-only mode.  Try adding:
```
afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY);
```

You might need to include the GPIO header for that function.  So if it says it can't find it, try also adding:
```
#include <libmaple/gpio.h>
```

to the top of your code file too.

dewhisna, 2019-09-25:
You can also use the more "standard" `wirish` defined functions of `disableDebugPorts()` or `enableDebugPorts()`, but I find their naming confusing, as both functions disable the full-JTAG, but the latter switches it to SW-only mode.  So I generally just use the lower-level `afio_cfg_debug_ports` function.

If I'm not mistaken, I believe either JTAG mode should free that pin for your SPI use.  JTAG doesn't get disabled by default in the core because it would interfere with users needing to use JTAG to recover a 'bricked' CPU or to do in-system debugging.

arnold-b, 2019-09-25:
Thank you for help. 
afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY);

was enough, now SPI3 also works. 
Next i need is SPI Slave with interrupt

closed 2019-09-25 by arnold-b