rogerclarkmelbourne/Arduino_STM32 #690: Wire missing readBytes method

TheKikGen, 2019-09-25:
Unlike the official Arduino I2C Wire library, the readBytes method, which allows you to read several received bytes at once, does not exist on the version ported to STM32F1.

I propose to add it, to be compatible with existing code coming from the AVR platform.
See a proposed version below affecting wire_slave.cpp and wire_slave.h.
Tested in one of my I2C app....works like a charm.

```
wire_salve.h

virtual size_t readBytes(uint8_t *data, const size_t quantity);

```

```
wire_slave.cpp

// must be called in:
// slave rx event callback
// or after requestFrom(address, numBytes)
size_t TwoWire::readBytes(uint8_t *data, const size_t quantity) {

  size_t qty = 0;

  // get bytes if some available
  if (rxBufferIndex < rxBufferLength) {

    // Do not ask more than available
    if ( quantity > rxBufferLength - rxBufferIndex ){
      qty = rxBufferLength - rxBufferIndex;
    } else qty = quantity;

    if ( qty && data != nullptr ) {
  		memcpy(data,&rxBuffer[rxBufferIndex], quantity);
  		rxBufferIndex += qty;
    }
  }

  return qty;

}
```

TheKikGen, 2019-09-25:
This could be added with the following pull request...
#679

dewhisna, 2019-09-26:
No ... `readBytes` is not missing.  Did you actually try it??  It's in the parent class for TwoWire, which is derived from `Stream`.  That's where the Wire class in the "official Arduino" library gets it (even their documentation for both [Stream](https://www.arduino.cc/en/Reference.StreamReadBytes) and [Wire](https://www.arduino.cc/en/reference/wire) says that it's derived from `Stream` in theirs too).

I just tested it by modifying one of the example sketches for WireSlave to call `readBytes` and it compiles just fine.  There's no need to add yet another implementation of the same function.  Now, it might be missing in the `Wire` library, since it has its own `WireBase` base class and doesn't appear to inherit from `Stream`, but it's certainly there in my reworked `WireSlave` library.  My new `WireSlave` implementation should be pretty true the "official Arduino" version and should inherit all of the `Stream` functionality.

TheKikGen, 2019-09-26:
Yes Donna. You'are right.   That was an error on my side still using a very specific version of wire.
I cloned again from your fork and all is working perfectly with the stream readBytes. I close that.
Nice job and thanks for the cleaning between wire and wire_slave !

closed 2019-09-26 by TheKikGen