rogerclarkmelbourne/Arduino_STM32 #692: Adafruit_ILI9341_STM::begin destroy SPI traffic

arnold-b, 2019-09-26:
Hello, 
i just try to use your sample. When i call tft.begin() 
the clock on SPI1 don´t work anymore. 

Finaly this call make it not usable:
mSPI.beginTransaction(SPISettings(_safe_freq, MSBFIRST, SPI_MODE0, DATA_SIZE_8BIT));

Main:
Adafruit_ILI9341_STM tft = Adafruit_ILI9341_STM(TFT_CS, TFT_DC, TFT_RST); // Use hardware SPI
//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST); // Using hardware SPI

#endif

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin PB1 as an output.
 //  afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY);// needed if using SPI3
  pinMode(PB1, OUTPUT);
  pinMode(PB0, OUTPUT);
  pinMode(PA1, OUTPUT);
  Serial.begin(115200);
  Serial.println("ILI9488 Test!");
  Serial1.begin(115200);
  Serial1.println("ILI9488 Test!");
 Serial2.begin(115200);
  Serial2.println("ILI9488 Test!"); 
  Serial3.begin(115200);
  Serial3.println("ILI9488 Test!");
  SPI.setModule(1);
  SPI.begin();
  #ifdef etft
  #else
  
  #endif
 tft.begin();
 
}

// the loop function runs over and over again forever
void loop() {
 
  Serial.println("Loop IO");
Serial.println("HWSPI0");
 Serial1.println("HWSPI1");
 Serial2.println("HWSPI2");
 Serial3.println("HWSPI3");
  digitalWrite(PB1, HIGH);   // turn the LED on (HIGH is the voltage level)
  digitalWrite(PB0, HIGH);  
 // digitalWrite(PA1, HIGH);  
  delay(200);     
  SPI.transfer(0xb4);// wait for a second
  digitalWrite(PB1, LOW);    // turn the LED off by making the voltage LOW
  digitalWrite(PB0, LOW);
 // digitalWrite(PA1, LOW);
  
  delay(200);              // wait for a second
}

without tft.begin
[image]

with tft.begin

[image]

stevstrong, 2019-09-26:
If you are using the Adafruit library, then you don't need to call SPI.begin() yourself in setup.

arnold-b, 2019-09-26:
Even i remove my SPI.begin(). It don´t solve my problem. 
I used just now the Library from https://github.com/jaretburkett/ILI9488

It works only if i disable #define SPI_HAS_TRANSACTION in spi.h

if i enable this then spi don´t work like in the upper pic

arnold-b, 2019-09-26:
Now i got it running with SPI_HAS_TRANSACTION. But i don´t know why ?
Its a bit slow compared to ESP32.  I have to agree, esp is faster and lib is diff.

Benchmark                Time (microseconds)
Screen fill              3623327
Text                     173686
Lines                    3639022
Horiz/Vert Lines         297853
Rectangles (outline)     169893
Rectangles (filled)      8996313
Circles (filled)         1146365
Circles (outline)        1560329
Triangles (outline)      701093
Triangles (filled)       3012778
Rounded rects (outline)  545762
Rounded rects (filled)   8830298

closed 2019-09-26 by arnold-b

stevstrong, 2019-09-26:
SPI_HAS_TRANSACTION is disabled by default in the SPI lib, see: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/a4007f57dfaf26a0e92191a261e312888b896995/STM32F1/libraries/SPI/src/SPI.h#L54
I suggest you leave it that way.
Instead, do SPI.begin() in setup.

On the other side, the lib from jaretburkett is a mess of different defines. You should know very precisely how to handle them instead of just playing around wildly with them.

arnold-b, 2019-09-26:
But i removed this in Library. With this the Library don´t work
#ifndef SPI_HAS_TRANSACTION

 //   SPI.setBitOrder(MSBFIRST);
 //   SPI.setDataMode(SPI_MODE0);

 //   SPI.setClockDivider(SPI_CLOCK_DIV2); // 8 MHz (full! speed!)
 
  
#endif